project-chip/connectedhomeip #29382: [ICD] Identify ICD Device in self-registration for auto-commissioning flow

yunhanw-google, 2023-09-21:
Read Descriptor Cluster and Check if ICDManagementCluster exists and further check if ICD feature is there or not from global feature map.

yunhanw-google, 2023-09-25:
@erjiaqing We also need to Read attribute  "current active mode" to allow dynamic switching of state based upon 
https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/7556#issuecomment-1734048250

mkardous-silabs, 2023-09-28:
@yunhanw-google @erjiaqing  I do not think the `current active mode` is necessary during the commissioning process. A Controller should register to the ICD no matter.

The `current active mode` once we have decided of its final form will only be used to determine if the controller should or should not buffer commands to the device.

PR #29427 (2023-10-30): [icd] Add steps for checking ICD during commissioning
Fixes #29382 

- Adds `kIcdDiscovery` stage during commissioning
- Adds `mIcdRegistrationStrategy` for commissioning params, with `kIgnore` and `kBeforeComplete`, `kBeforeComplete` should be used for self-registration or external controller registration.

Tests: WIP,  to be added in future PRs
closed 2023-10-30 by mergify[bot]