project-chip/connectedhomeip #29420: [Build] Built for esp32s2 target - ESP32-S2-SAOLA-1R (WROVER)

fernando-j-afonso-alb, 2023-09-23:
### Build issue(s)

I'm having problems building the lighting-app using esp32s2 as target

The board I'm using is a [ESP32-S2-SAOLA-1R (WROVER)](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-saola-1-v1.2.html)

```
...
/Users/fafonso/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/esp_wifi/libesp_wifi.a(esp_adapter.c.obj):(.iram1.11.literal+0x0): undefined reference to `ets_timer_arm'
/Users/fafonso/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/esp_wifi/libesp_wifi.a(esp_adapter.c.obj): in function `timer_arm_wrapper':
/Users/fafonso/esp/esp-idf/components/esp_wifi/esp32s2/esp_adapter.c:338: undefined reference to `ets_timer_arm'
/Users/fafonso/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/esp_coex/libesp_coex.a(esp_coex_adapter.c.obj):(.iram1.4.literal+0x0): undefined reference to `ets_timer_disarm'
/Users/fafonso/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/esp_coex/libesp_coex.a(esp_coex_adapter.c.obj):(.literal.esp_coex_common_timer_done_wrapper+0x0): undefined reference to `ets_timer_done'
/Users/fafonso/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/esp_coex/libesp_coex.a(esp_coex_adapter.c.obj):(.literal.esp_coex_common_timer_setfn_wrapper+0x0): undefined reference to `ets_timer_setfn'
/Users/fafonso/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/esp_coex/libesp_coex.a(esp_coex_adapter.c.obj):(.iram1.5.literal+0x0): undefined reference to `ets_timer_arm_us'
/Users/fafonso/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/esp_coex/libesp_coex.a(esp_coex_adapter.c.obj): in function `esp_coex_common_timer_disarm_wrapper':
/Users/fafonso/esp/esp-idf/components/esp_coex/esp32s2/esp_coex_adapter.c:107: undefined reference to `ets_timer_disarm'
/Users/fafonso/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/esp_coex/libesp_coex.a(esp_coex_adapter.c.obj): in function `esp_coex_common_timer_done_wrapper':
/Users/fafonso/esp/esp-idf/components/esp_coex/esp32s2/esp_coex_adapter.c:112: undefined reference to `ets_timer_done'
/Users/fafonso/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/esp_coex/libesp_coex.a(esp_coex_adapter.c.obj): in function `esp_coex_common_timer_setfn_wrapper':
/Users/fafonso/esp/esp-idf/components/esp_coex/esp32s2/esp_coex_adapter.c:117: undefined reference to `ets_timer_setfn'
/Users/fafonso/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/esp_coex/libesp_coex.a(esp_coex_adapter.c.obj): in function `esp_coex_common_timer_arm_us_wrapper':
/Users/fafonso/esp/esp-idf/components/esp_coex/esp32s2/esp_coex_adapter.c:122: undefined reference to `ets_timer_arm_us'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

Thank you.

### Platform

esp32

bzbarsky-apple, 2023-09-23:
@shubhamdp @dhrishi

aravindone, 2023-09-23:
@fernando-j-afonso-alb  please share full log of the terminal ? in .TXT format

fernando-j-afonso-alb, 2023-09-23:
I may found the problem, this device doesn't have BLE...

aravindone, 2023-09-26:
Hi @fernando-j-afonso-alb the devkit you have is WIFI SOC , and in the log 

> warning: the default selection DEVICE_TYPE_ESP32_H2_DEVKITC (undefined) of <choice> (defined at /Users/fafonso/Projects/AlticeLabs/Matter/connectedhomeip/examples/lighting-app/esp32/main/Kconfig.projbuild:23) is not contained in the choice
[1062/1064] Linking CXX executable chip-lighting-app.elf
FAILED: chip-lighting-app.elf 

so you have set-target for your board and try to build

jonkofee, 2023-11-12:
I have the same errors

dhrishi, 2024-10-25:
@shubhamdp