project-chip/connectedhomeip #29421: Bootstrap on Apple Silicon says protoc missing error

mustafayuce33, 2023-09-23:
### Reproduction steps

@woody-apple @bzbarsky-apple This will be the repeat of this issue https://github.com/project-chip/connectedhomeip/issues/14833. The issue still happens. 
As @dtodor said it still uses x86_4. Are you sure that this has been resolved?

 

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

https://github.com/project-chip/connectedhomeip/commits/master

### Platform

raspi

### Type

Core SDK Crash

bzbarsky-apple, 2023-09-23:
@mustafayuce33 Do you have Rosetta installed and turned on?

bzbarsky-apple, 2023-09-23:
Also, why is platform "raspi"?  What _actual_ hardware are you dealing with?  What exact commands are you running?  And please attach, not paste, the resulting logs.

mustafayuce33, 2023-09-23:
Thanks for quick reply. Do I need to install Rosetta? Then I will do that. I cross compile project on my mac m2 for raspberry pi. Below is the command I execute for building and resulting logs are attached.

./dynamic-bridge/third_party/connectedhomeip/scripts/examples/gn_build_example.sh ./out/cm4 target_cpu="arm" target_os="linux" sysroot="/Users/<username>/pi/sysroot" system_libdir="lib/arm-linux-gnueabihf" chip_config_network_layer_ble=false is_clang=true chip_mdns="platform"

[resultlog.txt](https://github.com/project-chip/connectedhomeip/files/12707100/resultlog.txt)

mustafayuce33, 2023-09-23:
After I install rosetta I've successfully built the project. I think you should add building instructions about installing `Rosetta` for Apple Silicons 

```
softwareupdate --install-rosetta
```

closed 2023-09-25 by mustafayuce33