project-chip/connectedhomeip #29422: [BUG] Operational State Start Command issue

shuqunx, 2023-09-25:
### Reproduction steps

If setting operational error = 0x1 (kUnableToStartOrResume) and operational-state = 0x3 (kError), Start command execute successful.

The condition in 
https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/operational-state-server/operational-state-server.cpp#L382 is incorrect, if OperationalState is not kRunning, HandleStartStateCallback() is called.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

master

### Platform

ameba

bzbarsky-apple, 2023-09-26:
>  if OperationalState is not kRunning, HandleStartStateCallback() is called.

The spec requirement is that if state is kRunning nothing happens.  Otherwise, the SDK has no idea whether the device can start or not, and it's up to the client application to make that decision, no?

I guess the cluster impl could "sniff" the current operational error if we are currently in error state... but 
`kUnableToStartOrResume` does not mean "can't start"; it could mean "can't resume".  So that's not enough to act on either, as far as I can tell.

@hicklin

shuqunx, 2023-09-27:
@bzbarsky-apple In TC-OPSTATE-2.2 Step 17 we have to set into a "state" where Start Command returns an ErrorStateID = 1, what could be the "state"?

bzbarsky-apple, 2023-09-27:
@shuqunx That depends on the device, right?  I don't see why there has to be such a state at all, in which case you should file an issue on the test plan, since it's assuming there is one.