project-chip/connectedhomeip #29425: [BUG] [Darwin] Failed to Open Commissioning Window

qawitherev, 2023-09-25:
### Reproduction steps

1. 'Device1' is commissioned to 'controller1'.
2. Now I want to 'controller2' to commission 'device1' using open commissioning window command.
3. Invoke open commission window command by the following steps.
4. Get discriminator (0 - 4096) from user (not sure if this from user or is automatically generated).
5. Generate random setup passcode by using `MTRSetupPayload.generateRandomSetupPasscode()`.
6. Use the `MTRBaseDevice.openCommissioningWindow()` passing in the **generated random passcode**, the **discriminator**, **timeout duration**, **callback queue** and **completion handler**. 
7. Inside the completion handler, can access the `MTRSetupPayload`. 
8. However the following error occurred.

[Device log]
[image]

[App log]
[image]

### Bug prevalence

Whenever i try to do this. 

### GitHub hash of the SDK that was being used

- 

### Platform

darwin

### Anything else?

The goal is to open commissioning window using ECM. By the end of this process, device will now inside two fabrics and can be controlled by controller of fabric 0x1 and 0x2. 

[ADDITIONAL QUESTIONS FOR CLARIFICATION]
1. Inside the `MTRBaseDevice.openCommissioningWindow()` completion handler, there's MTRSetupPayload that we can use. My question is, which one is the new setup passcode that we want show to user. Is it the result from `MTRSetupPayload.generateRandomSetupPasscode()` or any other `MTRSetupPayload` that isn't mentioned here?

PS: the opening window implementation is from the CHIP-Tool implementation. I'm not sure if there's any other way to open a commissioning window.

bzbarsky-apple, 2023-09-26:
@qawitherev What timeout duration did you use?  Ideally the device would log why it considers the command invalid, but in practice the 
```
Failed to open commissioning window. Global status 0x85
```
log means the provided timeout was outside the spec-required range of 180-900 seconds (3-15 minutes).

> My question is, which one is the new setup passcode that we want show to user. Is it the result from MTRSetupPayload.generateRandomSetupPasscode() or any other MTRSetupPayload that isn't mentioned here?

You want to show the user a serialization of the MTRSetupPayload that's passed to the completion handler, as either a QR code or a numeric code.

qawitherev, 2023-09-29:
Timeout is 180 second, noted. Controller (app) is not notified when device is commissioned to a new admin is it? Its ok tho. Closing this one. Big thanks my guy!

closed 2023-09-29 by qawitherev

bzbarsky-apple, 2023-09-29:
> Controller (app) is not notified when device is commissioned to a new admin is it?

It is if it subscribes to the Fabrics attribute on the device...