project-chip/connectedhomeip #29434: [Chef][Bug?]  Unable to create a switch type with Switch Generic for up and down paddles

jvmahon, 2023-09-25:
### Feature description

@bzbarsky-apple <- Tagged since this concerns an example type that you last updated.

**Summary**
This concerns an inability to use the light-switch-app example files with Chef to create a virtual switch device with support for "taps" on the paddles.

**What I was trying to do**
I'm trying to use Chef to create what I think is a pretty basic device - a dimmer switch type with support for up to 5 taps, hold, release, on the upper and lower paddles.  Basically, I'm trying to replicate the functionality on a Z-Wave Dimmer that supports Z-Wave central scene commands.

**My approach**
My starting point for this is to use the .zap and .matter files from the light switch app (https://github.com/project-chip/connectedhomeip/tree/master/examples/light-switch-app/light-switch-common)

I then copied these files to the Chef devices directory and compiled them on Chef as a "virtual" device on a raspberry pi 4. Output to the -t linux type for testing - i.e., no physical device, just the pi.  

I get everything to compile and run and commission to iOS Home (version 17.0.2, iphone 15). 

**What went wrong**
Following commissioning, a new device does appear in iOS home, but, I only see a single "tap" generic switch device.  The endpoint for the on / off light switch itself doesn't appear.

**Questions / Followup**
I'm trying to determine if this is user error, an iOS error, or other.  Basic questions:
1. has anyone gotten a switch or dimmer example with generic switch endponts added for taps/holds/releases working, and does it show properly in iOS (and Google Home, etc.)?
2. Is anyone willing to try and replicate this - i.e., can you get the light-switch-app working on a pi and commission to iOS to see if it works for someone else?

**New Example for Chef**
I would like to see a "permanent" addition to Chef that includes a working "dimmer" example with support for paddle taps  (up to 5 taps, hold, release).(Endpoint 0 -> Root, Endpoint 1 -> Dimmer 0x0104 type, Endpoint 2 -> Generic Switch 0x000F, Upper Paddle, Endpoint 3 -> Generic Switch 0x000F, Lower Paddle) be added to Chef. Preferably, a "complete" example - e.g., how to use Zap to label the endpoints 2/3 ("Upper Paddle", "Lower Paddle") - I can't figure this out either.

Basically, a example dimmer device that replicates what you commonly find in the Z-Wave market dimmer market.

### Platform

all

bzbarsky-apple, 2023-09-26:
> the endpoint for the on / off light switch itself doesn't appear.

@jvmahon I believe Apple Home supports "Generic Switch" devices but not "On/Off Switch" devices.  Past that, I am not clear on what the difference is between what you were expecting Apple Home behavior to be and what it actually was.

The right place to report Apple Home issues is via Feedback Assistant.  Please file a ticket there and drop the ticket ID here so I can point people to it.

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

jvmahon, 2023-09-26:
iOS definitely supports On/Off switches (0x0103) - there are several on the market, including some by Tapo which I regularly use, so that's not it.  I've also tried using the Dimmer type (0x104).

I think the issue may be with the .zap example files (I noticed this after my original post)

 Section 2.1.5 of the Matter Device Library Specification Version 1.1 states:
[image]

However, in each of the example .zap files, when you open the Zap file with the Zap tool, the PartsList for endpoint 0 is always empty.  That seems to be contrary to the specification and may be why it fails to work properly.  (I'm new to this, so it may also be a problem with my understanding).

Is my spec understanding correct (and, accordingly, my understanding that the chef example files don't conform)? If so, I could try to do a PR to correct, but don't want to touch that in case I have a misunderstanding.

bzbarsky-apple, 2023-09-27:
PartsList is determined at runtime by the Matter SDK; it's not configured via ZAP.  That should not be the issue here.

jvmahon, 2023-09-29:
@bzbarsky-apple 
I wanted to follow up and thank you for your help. I was able to get the Generic Switch added. It turns out that Zap did not include the Descriptor cluster and when I added that manually, it seemed to work.

There is a significant user interface error with that feature on iOS that I found which I will be reporting to Apple. You mentioned that once I report to let you know and you would also relay that, so I'm keeping this open until I report that. Thanks.

closed 2023-10-19 by jvmahon

jvmahon, 2024-01-13:
> The right place to report Apple Home issues is via Feedback Assistant. Please file a ticket there and drop the ticket ID here so I can point people to it.

I just submitted feedback to Apple on a battery reporting issue. It seems iOS does not report battery % when a device properly conveys the information via a Power Source endpoint, but will do so if the device adds the power cluster to another endpoint (even though Power Source is not designated as an optional cluster on the other endpoint). 

You previously noted it may be helpful to draw your attention to reports entered in Apple Feedback so you can direct them to the right person.   So, in case there is anything else you can do to get this noticed:   https://feedbackassistant.apple.com/feedback/13537833

jvmahon, 2024-01-13:
And this: https://feedbackassistant.apple.com/feedback/13487464

bzbarsky-apple, 2024-01-18:
>  https://feedbackassistant.apple.com/feedback/13537833

Prodded some people.

>  https://feedbackassistant.apple.com/feedback/13487464

Likewise.

jvmahon, 2024-01-18:
Greatly appreciated. Thank you!

jvmahon, 2024-01-29:
Two more for you (these are strictly Apple Matter related, not the SDK, so up to you whether they are forwarded). I haven't seen any response to the items I put in Apple Feedback so maybe they get more notice when you forward them.

1. Apple's new tvOS beta incorrectly processes the firmware number field for matter devices:
https://feedbackassistant.apple.com/feedback/13565944

2. Apple iOS Home labels all Generic Switch endpoints as "Button1". Distinct labels should be used (e.g., based on FixedLabel or TagList)
https://feedbackassistant.apple.com/feedback/13566249

bzbarsky-apple, 2024-02-01:
Prodded some people.

jvmahon, 2024-07-11:
> Prodded some people.

@bzbarsky-apple   I am hoping you can "prod" the right people for a error introduced in the iOS 18 beta regarding the handling of Matter Generic Switch / button tapping events.

I've submitted a report here: https://feedbackassistant.apple.com/feedback/14275530   (but my experience is the feedback can get overlooked so  would appreciate your help).

Here's a summary of what I described in the feedback report.

iOS 18 Beta has introduced an error in the handling of button tap detection for Matter devices. Specifically, this refers to the processing of Matter Generic Switch (Matter Device ID 0x000F, Matter Cluster ID 0x003B).

1. Apple iOS allows a device to report button tapping events. Specifically, (i) Single Press, (ii) Double Press, and (iii) Long Press.
2. Matter Devices can report these events, but can also generate events for additional presses (e.g., triple press, 4X press, etc.
3. In iOS 17.5, TVOS 17.5, if a Matter device generated a single, double, or Long press, it was properly detected by iOS. That's expected
4. In iOS 17.5, tvOS 17.5, if a Matter device generated more than double-press, the event was ignored.   This is also expected.
5. In iOS 18 beta 3, however, the behavior described in # 4 was changed. Now, if a Matter device generates more than double-press (e.g., 3 presses, 4 presses, etc.) iOS now treats it as a double press. This is the error.
6. What should happen is that if a 3X press, 4X press, etc. is generated, iOS should ignore it ( as it used to do in iOS 17.5) rather than treat every press count greater than 2 as if it were a double press.
7. Ideally, iOS should be expanded to properly recognize these additional press counts greater than 2, but it is better to ignore these greater-than-2 press counts rather than treat them as double presses.

bzbarsky-apple, 2024-07-12:
@jvmahon That ticket does not seem to have sysdiagnoses attached for the device Home is running on?

Could you follow the directions at https://github.com/project-chip/connectedhomeip/issues/29434#issuecomment-1735817554 to create a feedback assistant ticket with the relevant logs?  Or were you following those and something went wrong?

jvmahon, 2024-07-12:
@bzbarsky-apple 
I did have problems with the syslog, but I have now gone back and added the syslog files.

As a bit of additional information, the specific device at issue is this one: https://inovelli.com/collections/inovelli-white-series/products/thread-matter-white-series-smart-2-1-on-off-dimmer-switch

If your contacts need a device to confirm and fix the issue, I could have one sent.

aajain-com, 2024-07-16:
@jvmahon will take a look at the issue shortly and respond on the FB ticket directly.

jvmahon, 2024-07-17:
> @jvmahon will take a look at the issue shortly and respond on the FB ticket directly.

@aajain-com - I'd also suggest the best way to solve the issue is for you to actually have one of the Invelli switches for you to test  Inovelli will send you one if you can send a mailing address. Let me know if they can send one to you (and where).

aajain-com, 2024-07-18:
@jvmahon  please attach tvOS 17.5 to the feedback ticket.

jvmahon, 2024-07-18:
@aajain-com 
Thanks Anshul,

Its not possible for me to attach the 17.5 sysdiag. The iOS/tvOS 18 beta does not allow downgrading back to tvOS 17.5.

I did attach the sysdiags for the iOS / tvOS 18 beta.

I do think the easiest solution would be to have a device sent to you so you can observe the issue yourself. Please let me know if there is an address to send it to.

Thank you for your help.