project-chip/connectedhomeip #29439: [BUG] Custom cluster's Read and Subscribe interactions doesn't work

8TRobi8, 2023-09-25:
### Reproduction steps

I have created a custom cluster that is required for our setup, so I've made the XML file that describes the custom cluster, then added the new cluster to the required endpoint in the ZAP file, and then regenerated the ZAP-generated files.

The custom cluster has one command and one attribute. The command handling works fine, the controller receives the command and the payload correctly. However, the attribute handling does not work well. I tried to read the attribute value with chip-tool, but I always got an INVALID_ACTION response. In the controller's log, I see `chip[EM]: Dropping unexpected message of type 0x1 with protocolId (0, 1) and MessageCounter:219703856 on exchange 24614r`.

The custom cluster's XML content:
```
<?xml version="1.0"?>
<configurator>
  <domain name="General"/>

  <cluster>
    <name>TestCluster</name>
    <domain>Home Automation</domain>
    <code>0xFF00</code>
    <define>TEST_CLUSTER</define>
    <description>TestCluster description</description>
    <globalAttribute side="either" code="0xFFFD" value="1"/>

    <attribute side="server" code="0x0000" define="STATUS" type="CHAR_STRING" length="128" reportable="true">Status</attribute>

    <command source="client" code="0x00" name="Change" optional="false">
        <description>Description of the Change command</description>
        <arg name="message" type="CHAR_STRING" length="64"/>
    </command>

  </cluster>
</configurator>
```

Could someone help me out, how can I solve this problem?

### Bug prevalence

always

### GitHub hash of the SDK that was being used

4088a77

### Platform

core

bzbarsky-apple, 2023-09-26:
>     `<code>0xFF00</code>`

That's not a valid cluster id, per spec.  That's why it's failing.  Presumably on the `IsValidClusterId()` check under `ProcessAttributePaths`.

It looks like CommandHandler does not do such a check (arguably a bug), which is why your command works.

The valid cluster id ranges are:

* 0x00000000 - 0x00007FFFF -- standard-defined cluster
* 0x0001ABCD - 0xFFF4DEFG -- manufacturer-defined cluster, so long as ABCD >= FC00 and DEFG <= FFFE.  The first two bytes are the manufacturer id.

As in, you need to add your manufacturer id to the cluster code.

bzbarsky-apple, 2023-09-26:
Also, note that manufacturer IDs 0xFFF1-0xFFF4 are reserved for testing.

bzbarsky-apple, 2023-09-26:
Note https://github.com/project-chip/connectedhomeip/pull/29461 for fixing writes/commands to also do the checks.

BryanH1128, 2023-10-19:
May I know which Chip-tool commands can be used to read Custom Clusters?
We have added custom clusters to my Matter device but cannot find out which command we should use on the Matter Controller.

bzbarsky-apple, 2023-10-21:
> May I know which Chip-tool commands can be used to read Custom Clusters?

`chip-tool any ....`

BryanH1128, 2023-10-23:
> > May I know which Chip-tool commands can be used to read Custom Clusters?
> 
> `chip-tool any ....`
Ah... We looked at the commands list a lot of times...
Didn't notice that the "any" can be used by setting the IDs... 
Thanks a lot.

BryanH1128, 2023-11-06:
> > May I know which Chip-tool commands can be used to read Custom Clusters?
> 
> `chip-tool any ....`

Hi Boris,

I highly appreciate your help that I can now use the 'any' command to write the custom cluster.
"./chip-tool any write-by-id 6 0xfff18000 1 11 4"
[image]

However, it is weird that I cannot read the same ID back.
"./chip-tool any read-by-id 6 0xfff18000 11 4"
[image]

Do you have any ideas about this issue?
Many thanks for your help.

bzbarsky-apple, 2023-11-13:
@BryanH1128 `0xfff18000` is not a valid attribute ID.  Per spec, attribute IDs for non-global attributes must have the low 4 bytes in the range 0x0000 to 0x4FFF.

It looks like the server in this case does not enforce that for writes (though I suspect tip does) but does enforce it for reads.

BryanH1128, 2023-11-14:
> @BryanH1128 `0xfff18000` is not a valid attribute ID. Per spec, attribute IDs for non-global attributes must have the low 4 bytes in the range 0x0000 to 0x4FFF.
> 
> It looks like the server in this case does not enforce that for writes (though I suspect tip does) but does enforce it for reads.

Hi Boris,
I highly appreciate your help.
After changing the ID, it is working for read and write now. :')
Thanks a lot.

bzbarsky-apple, 2023-11-16:
@BryanH1128 Glad to hear it work!