project-chip/connectedhomeip #29449: [BUG] Cannot read Power Source Configuration cluster's Sources attribute - Matter Bridge

xvortep, 2023-09-26:
### Reproduction steps

Greetings,
The following scenarios are tried:
# First scenario
1. On EP0, where the root node resides, I added the Power Source Configuration Cluster with ZAP.
2. On EP2, which is used to define all the clusters used by bridged devices, I added the Power Source Cluster with ZAP.
3. Dynamic attributes, clusters and endpoints for a Contact Sensor device with battery level are added.
4. The Contact Sensor Device is added correctly and the Boolean State can be read from it, as well as the Battery Level using chip-tool.
5. Reading Sources attribute from EP0 throws the following error:

```
[1695636734.460217][14263:14265] CHIP:TOO: Response Failure: IM Error 0x00000501: General error: 0x01 (FAILURE)
[1695636734.460251][14263:14265] CHIP:TOO: Run command failure: IM Error 0x00000501: General error: 0x01 (FAILURE)
```
# Second scenario
1. On EP2 both Power Source Configuration and Power Source clusters are added with ZAP.
2. Both Clusters are defined with DECLARE_DYNAMIC... and are added to dynamic endpoints.
3. Battery Level can be read successfully like in the first scenario.
4. Reading Sources attribute from EP2 throws the following error:
`UNSUPPORTED_READ`

As I understand, the Power Source Configuration cluster should reside on the root node, and should indicate which endpoints contain Power Source clusters, so the controller should be able to read the Sources attribute.

Platform: Linux

### Bug prevalence

Every Time

### GitHub hash of the SDK that was being used

d8480f1859

### Platform

core

bzbarsky-apple, 2023-09-26:
> Reading Sources attribute from EP0 throws the following error:

I assume EP2 (with the dummy power source cluster) is disabled?

What do the server logs show?  Why did it decide to return FAILURE?  Can you look at whether `PowerSourceConfigurationAttrAccess::Read` is being called on the server, and if so why it returns an error?

bzbarsky-apple, 2023-09-26:
> Reading Sources attribute from EP2 throws the following error

Putting Power Source Configuration on EP2, which is then disabled (right?) doesn't really make sense.

bzbarsky-apple, 2023-09-26:
Note that if you are dealing with power sources for the _bridged_ devices, then power source configuration should go on the bridge node endpoints, not in EP0; EP0 is for describing the power sources for the bridge itself.

xvortep, 2023-09-27:
> I assume EP2 (with the dummy power source cluster) is disabled?
>
> What do the server logs show? Why did it decide to return FAILURE? Can you look at whether `PowerSourceConfigurationAttrAccess::Read` [...]
Yes, it's disabled.
So, because the Power Source Configuration cluster is on EP2, and it is declared as dynamic, should I have to handle the Sources attribute or the core?

If I have to handle the attribute via emberAfExternalAttributeRead myself, how should I return the list, as a ZCLCharString or as something else?

And what is the point of the Power Source Configuration Cluster, if a device points to itself via Sources Attribute?

Thank you in advance

bzbarsky-apple, 2023-09-27:
> If I have to handle the attribute via emberAfExternalAttributeRead myself

You can't handle a list-typed attribute via emberAfExternalAttributeRead.  It _has_ to be handled via AttributeAccessInterface.

> And what is the point of the Power Source Configuration Cluster, if a device points to itself via Sources Attribute?

I'm not sure I follow....

That said: there is not that much point to the Power Source Configuration cluster, which is why it's being deprecated in Matter 1.2.  It was created before people realized there were other options for discovering what endpoints have Power Source clusters on them.

xvortep, 2023-09-28:
> That said: there is not that much point to the Power Source Configuration cluster, which is why it's being deprecated in Matter 1.2. It was created before people realized there were other options for discovering what endpoints have Power Source clusters on them.

Okay, that simplifies it. Thank you.

Just so I don't open up a new issue (or if you think it deserves a new issue, I will do so):

Do you have any plans for providing some sort of documentation about bridging devices? The example shows the most trivial behavior and there is no explanation of how the attributes like feature map should be handled. Also, when adding a dynamic endpoint with certain clusters, when, and when should I not call it's clusterInitCallback? On top of that, isn't it a good idea to leave that to the sdk itself to handle?

Thank you.

bzbarsky-apple, 2023-09-28:
> there is no explanation of how the attributes like feature map should be handled

Well... "per spec", right?  The feature map will depend on the actual features implemented.

> Also, when adding a dynamic endpoint with certain clusters, when, and when should I not call it's clusterInitCallback?

If the callback is declared in the data defining the endpoint, it will be called automatically when the endpoint is enabled....  Note that the DYNAMIC_* macros don't have a way to specify the callback, as I recall, so for clusters that have one you need to unroll the macros and add it to the result manually...

And yes, it would be good if the SDK knew about this sort of thing automatically somehow; right now it's kind of tied to the ZAP codegen and the dynamic stuff is all about constructing at runtime the sort of data structure ZAP generates.  But I am hoping to make it better sometime in the next few months....

xvortep, 2023-09-29:
> > there is no explanation of how the attributes like feature map should be handled
> 
> Well... "per spec", right? The feature map will depend on the actual features implemented.

I mean, there are several ways to assign a value to an attribute:
* with ZAP UI
* by passing a value to the DECLARE_DYNAMIC_ATTRIBUTE
* via emberAfExternalAttributeRead

> If the callback is declared in the data defining the endpoint, it will be called automatically when the endpoint is enabled.... Note that the DYNAMIC_* macros don't have a way to specify the [...]

I guess it should be added to the EmberAfCluster::functions array, right?
I see there is a getter emberAfFindClusterFunction which returns a specific function from the aforementioned array based on the bitmask provided. A matching setter function (or a macro, so it would be added to the cluster in build-time) would be quite nice.
This could also be solved by extending ZAP to be able to define dynamic clusters etc.

bzbarsky-apple, 2023-09-29:
> I mean, there are several ways to assign a value to an attribute:

The ones you list are mutually exclusive options (there's also AttributeAccessInterface, which you did not list); which one you use depends on whether this is a static or dynamic endpoint and on where you want the value to actually be stored in memory.

> I guess it should be added to the EmberAfCluster::functions array, right?

Yes.  And the right flag should be set for the cluster to indicate it's in the array.

xvortep, 2023-10-05:
> > I mean, there are several ways to assign a value to an attribute:
> 
> The ones you list are mutually exclusive options 

Are they? It doesn't matter whether or not the value is passed by DECLARE_DYNAMIC_ATTRIBUTE, if the attribute is defined this way, it will be read using emberAfExternalAttributeRead

>(there's also AttributeAccessInterface, which you did not list); which one you use depends on whether this is a static or dynamic >endpoint and on where you want the value to actually be stored in memory.

That is what I'm talking about - there are different methods but there is no explanation of how and when should the one be used.

bzbarsky-apple, 2023-10-05:
> It doesn't matter whether or not the value is passed by DECLARE_DYNAMIC_ATTRIBUTE, if the attribute is defined this way, it will be read using emberAfExternalAttributeRead

`DECLARE_DYNAMIC_ATTRIBUTE` does not have a way to "pass a value"....  If you use DECLARE_DYNAMIC_ATTRIBUTE, it's an external attribute, period:
```
#define DECLARE_DYNAMIC_ATTRIBUTE(attId, attType, attSizeBytes, attrMask)                                                          \
    {                                                                                                                              \
        ZAP_EMPTY_DEFAULT(), attId, attSizeBytes, ZAP_TYPE(attType), attrMask | ZAP_ATTRIBUTE_MASK(EXTERNAL_STORAGE)               \
    }
```

> there are different methods but there is no explanation of how and when should the one be used.

I agree that the documentation is quite lacking, yes.

bzbarsky-apple, 2023-10-07:
@xvortep I don't know of specific plans to change how things are built, though the "integrate better with CMake" ask is pretty common.  I personally am not working on anything in that space.

In general, a big chunk of the SDK is in fact already built into libCHIP.a.  This does not include any cluster implementations or anything else that uses zap config, which is precisely the pain point at issue here, sounds like.

xvortep, 2023-10-12:
All right, thank you.

I've encountered a problem when updating the SDK. When I move to a different commit, then source bootstrap, the tests don't pass.
The only bulletproof way I've found for updating the SDK is to follow this flow:
git checkout #hash
git clean -xdf
git submodule update --init
source scripts/bootstrap.sh

However, this method is very slow, bootstrap and submodule update usually take about half to a full hour.
Is there any other way that does this quicker?

bzbarsky-apple, 2023-10-12:
What I normally do for incremental updates is:
```
 git checkout #hash
 git submodule update --init
```
If re-bootstrap is needed (which is not always, but I guess if you are updating across a long time it might be), then I just run `source scripts/bootstrap.sh` and that typically works.

In very rare cases I've had to `rm -r .environment` to clear out the bootstrap state.

And of course on initial checkout, you want `scripts/checkout_submodules.py` possibly with the relevant platform args, not just a full "pull all the submodules", because that _is_ slow.