project-chip/connectedhomeip #29450: [BUG] [ICD] Failed to switch to Idle mode if the device fails to post `kChipMsgRxEventHandled` event

wqx6, 2023-09-26:
### Reproduction steps

We will post a event `kChipMsgRxEventHandled` when an exchange context sends a message that expects response. This event will increase the `expectedMsgCount` in `ICDEventManager` by one. And another event `kChipMsgRxEventHandled` will be post when the device receives a response or the ResponseTimer is triggered. This event will decrease the `expectedMsgCount` in `ICDEventManager` by one.

If the `kChipMsgRxEventHandled` event is post but the device fails to post `kChipMsgRxEventHandled` event. The  `expectedMsgCount` in `ICDEventManager` will never be decreased to 0 and the device will never switch to Idle mode if we don't reboot the device.

### Bug prevalence

Sometimes

### GitHub hash of the SDK that was being used

6bb0b716bc19c0e52d25a1812bc729ca202123b5

### Platform

core

bzbarsky-apple, 2023-09-26:
We really need to not be using events for this and use some sort of actual notifications that cannot fail....  @jmartinez-silabs

jmartinez-silabs, 2023-09-26:
@bzbarsky-apple A big part of Matter SDK is constructed around those post-events and the chip event queues (chip task). A well-configured project should have a queue large enough and a processing speed fast enough to handle all the events. In those cases, the post does not fail.

The whole ICD event Manager is based on this principle currently. One of the reasons for using that mechanism was to remove circular dependency on the matter core and the ICD which is in the app layer.

I can start thinking of ways to refactor the ICD event Manager and us a different mechanism. I am open to suggestions.

If our assumption is that the post-event can fail under normal circumstances, then other parts of matter need to be reviewed

bzbarsky-apple, 2023-09-27:
> If our assumption is that the post-event can fail under normal circumstances

In practice we have seen a _lot_ of reports of this happening, because just minimal-mdns on its own can completely fill the queue....

jmartinez-silabs, 2023-09-28:
Any suggestion for another notification mechanism that wouldn't create a circular dependency between `src/app` and src/messaging?

I could maybe create a task for the ICD server and use task notification or maybe a counting semaphores. But I think this would increase code size again, and calls to the notifications would cause dependency issues in src/messaging

bzbarsky-apple, 2023-09-28:
The simplest way to do the dependency management seems to be:

1. Define a class that can get notifications for ICD stuff.  Make sure this header does not depend on anything except core and support.  Put this header into its own source set.  It can live under src/app or whatever, but the important thing is that this source set depend on nothing except core and support.
2. Have messaging depend on that source set.
3. Have whatever has the _implementation_ for that header depend on the source set with the header.

PR #29546 (2023-10-10): [ICD-server] Replace the event/queue based logic with a notifier
fixes #29450

- Replace the ICDEventManager and its event queue logic with the ICDNotifier (Subject/observer pattern). This removes the risk of the event queue being full and failing to post events which can get the ICD stuck in active mode.
- Rework the messaging Keep active requirements. Now it is purely tied to Open ExchangeContexts.
- Messages sent and received now trigger a "NetworkActivity" which kicks an active Threshold
- Tweak some ICDManager logic to account for SIT with active interval and active threshold of 0 ms allo [... cut]
closed 2023-10-10 by mergify[bot]