project-chip/connectedhomeip #29453: [Feature] MAD Switch cannot control Light device after Matter hub (Raspberry Pi 4B) is turn off

chenek, 2023-09-26:
### Feature description

...
I build and run Matter Over Thread Light and Switch on Silabs BRD4187C. The MAD Light and Switch join Matter hub (Raspberry Pi 4B) and binding is done on Switch so MAD switch can toggle MAD light when Matter hub (Raspberry Pi 4B) is turn on. When I turn off Matter hub (Raspberry Pi 4B) and wait a while (see "SRP Client was stopped, because current server is no longer detected" on RTT viewer output of MAD light/switch devices), MAD switch cannot control light. I suppose Matter over Thread is a mesh network and should not stop working when any single point of failure including Matter hub. Am I correct on this point? If so, why current open source implementation doesn't work in this way?

### Platform

efr32, raspi

bzbarsky-apple, 2023-09-26:
Matter over Thread is a mesh network, but relies on DNS-SD, which uses SRP over Thread.

Now you can have multiple SRP servers in your mesh, and then it will be somewhat resilient to any one of them going down.  But if you take down all the SRP servers (and it sounds like you only had one in your case), then DNS-SD is not going to work.

chenek, 2023-09-26:
So, how can we run multiple SRP server in the same Matter network? Can we run it on MAD?

bzbarsky-apple, 2023-09-27:
> how can we run multiple SRP server in the same Matter network? 

That's a Thread question, not a Matter question....

> Can we run it on MAD?

What is "MAD"?

chenek, 2023-09-27:
MAD stands for Matter Accessory Device. I just mean to know what is the alternative when OTBR/SRP server is down so the Matter over Thread mesh can still work.

bzbarsky-apple, 2023-09-28:
> what is the alternative when OTBR/SRP server is down so the Matter over Thread mesh can still work

You need to have more than one SRP server on the mesh if you want it to be resilient to one SRP server going down.  That could be multiple OTBRs, or some other devices that implement an SRP server.

chenek, 2023-09-28:
“Reliability — Self-healing mesh networking, with no single point of failure, and spread-spectrum techniques to provide immunity to interference” is claimed by https://openthread.io/guides/thread-primer?hl=en but the reality is not as claimed…
Anyway, I appreciate your help on clarifying this.

bzbarsky-apple, 2023-09-28:
Look, if you have a single router-capable device in your Thread mesh (and all the other things are sleepy end devices) and that one thing goes down, then things won't work either, right?  If you want a service provided on the mesh, but you only set up one device that can provide the service, then if that device goes down the service is not going to be available.  The right answer is to set up multiple devices that can provide the service.

To get very concrete for a second: if you have an Apple Home with multiple Homepods or AppleTVs on the Thread mesh, then things will work correctly if one of those goes down, because a different one takes over the SRP server function.

chenek, 2023-09-29:
Agree this can be solved by applying multiple OTBR but it’s not cost effective for end customers. In comparing to legacy Zigbee protocol, it would maintain local mesh working without coordinator/trust center which might be what end customers prefer. Anyway, that’s just personal opinion.