project-chip/connectedhomeip #29455: [Tooling]

ThinSnorlax, 2023-09-26:
### Reproduction steps / Feature

Hi 

I build "chip-cert"(https://github.com/project-chip/connectedhomeip/tree/master/src/tools/chip-cert) in ubuntu22;
The command is below of:
```
sudo apt update
sudo apt install libglib2.0-dev
. ./scripts/activate.sh
gn gen out/host
ninja -C out/host chip-cert
```
I can't run command "ninja -C out/host chip-cert";
Error:
```
Admin:~/environment/connectedhomeip (master) $ ninja -C out/host chip-cert
ninja: Entering directory `out/host'
[57/244] c++ obj/src/ble/libBleLayer.BleError.cpp.o
FAILED: obj/src/ble/libBleLayer.BleError.cpp.o 
g++ -MMD -MF obj/src/ble/libBleLayer.BleError.cpp.o.d -Wconversion -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -fmacro-prefix-map=../../= -std=gnu++17 -fno-rtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -I../../src/include -I../../src -Igen/include -I../../zzz_generated/app-common -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/include -c ../../src/ble/BleError.cpp -o obj/src/ble/libBleLayer.BleError.cpp.o
In file included from ../../src/lib/support/SetupDiscriminator.h:26,
                 from ../../src/ble/BleLayer.h:57,
                 from ../../src/ble/BleError.cpp:31:
../../src/lib/support/CodeUtils.h:130:10: fatal error: nlassert.h: No such file or directory
  130 | #include <nlassert.h>
      |          ^~~~~~~~~~~~
compilation terminated.
[58/244] c++ obj/zzz_generated/app-common/app-common/zap-generated/libClusterObjects.cluster-objects.cpp.o
FAILED: obj/zzz_generated/app-common/app-common/zap-generated/libClusterObjects.cluster-objects.cpp.o 
g++ -MMD -MF obj/zzz_generated/app-common/app-common/zap-generated/libClusterObjects.cluster-objects.cpp.o.d -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -fmacro-prefix-map=../../= -std=gnu++17 -fno-rtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -I../../src/include -I../../src -Igen/include -I../../zzz_generated/app-common -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/include -c ../../zzz_generated/app-common/app-common/zap-generated/cluster-objects.cpp -o obj/zzz_generated/app-common/app-common/zap-generated/libClusterObjects.cluster-objects.cpp.o
In file included from ../../src/system/SystemLayer.h:33,
                 from ../../src/lib/core/CHIPCore.h:30,
                 from ../../src/lib/core/Optional.h:28,
                 from ../../src/app/ConcreteAttributePath.h:23,
                 from ../../zzz_generated/app-common/app-common/zap-generated/cluster-objects.h:27,
                 from ../../zzz_generated/app-common/app-common/zap-generated/cluster-objects.cpp:20:
../../src/lib/support/CodeUtils.h:130:10: fatal error: nlassert.h: No such file or directory
  130 | #include <nlassert.h>
      |          ^~~~~~~~~~~~
compilation terminated.
[59/244] c++ obj/src/ble/libBleLayer.BLEEndPoint.cpp.o
FAILED: obj/src/ble/libBleLayer.BLEEndPoint.cpp.o 
g++ -MMD -MF obj/src/ble/libBleLayer.BLEEndPoint.cpp.o.d -Wconversion -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -fmacro-prefix-map=../../= -std=gnu++17 -fno-rtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -I../../src/include -I../../src -Igen/include -I../../zzz_generated/app-common -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/include -c ../../src/ble/BLEEndPoint.cpp -o obj/src/ble/libBleLayer.BLEEndPoint.cpp.o
In file included from ../../src/ble/BLEEndPoint.cpp:41:
../../src/lib/support/CodeUtils.h:130:10: fatal error: nlassert.h: No such file or directory
  130 | #include <nlassert.h>
      |          ^~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
```

like the error report, I can't find the nlassert.h in Path "/src/lib/support/"
What should I do for fixed the error?
I want to test my DAC and my environment is build on AWS EC2;

### Platform

other

### Platform Version(s)

ubuntu22

### Type

Platform validated

bzbarsky-apple, 2023-09-26:
>  What should I do for fixed the error?

Did you check out submodules?  I suggest running `scripts/checkout_submodules.py` to check out the minimal set for your OS.

ThinSnorlax, 2023-09-27:
> > What should I do for fixed the error?
> 
> Did you check out submodules? I suggest running `scripts/checkout_submodules.py` to check out the minimal set for your OS.

Hi 

I have run this command  :
```
Admin:~/environment/connectedhomeip (master) $ python3 scripts/checkout_submodules.py
Checking out: nlassert, nlio, nlunit-test, mbedtls, qrcode, pigweed, openthread, nanopb, third_party/pybind11/repo, third_party/jsoncpp/repo, editline, third_party/boringssl/repo/src, third_party/st/STM32CubeWB
```
Does it mean I need install nlassert, nlio, nlunit-test, mbedtls, qrcode, pigweed, openthread, nanopb for my environment?

bzbarsky-apple, 2023-09-27:
> Does it mean I need install nlassert, nlio, nlunit-test, mbedtls, qrcode, pigweed, openthread, nanopb for my environment?

It means that their code was grabbed by the checkout_submodules.py script.  In particular, now nlassert.h should exist, as part of the nlassert submodule.