project-chip/connectedhomeip #29465: [IM]When testing with group write, and endpointId is set with actual value, the server has not yet complained the error

yunhanw-google, 2023-09-27:
Based upon feedback from
Any EndpointId can send to Controllee. I think controllee is ignored this endpoint ID.
[7:31](https://csamembers.slack.com/archives/D03QTDMAA5R/p1695781895692419)
[1695781738.670899][29270:29270] CHIP:IM: Received Write request
[1695781738.670917][29270:29270] CHIP:DMG: IM WH moving to [Initialized]
[1695781738.670961][29270:29270] CHIP:DMG: WriteRequestMessage =
[1695781738.670982][29270:29270] CHIP:DMG: {
[1695781738.670999][29270:29270] CHIP:DMG:      suppressResponse = false,
[1695781738.671020][29270:29270] CHIP:DMG:      timedRequest = false,
[1695781738.671038][29270:29270] CHIP:DMG:      AttributeDataIBs =
[1695781738.671069][29270:29270] CHIP:DMG:      [
[1695781738.671087][29270:29270] CHIP:DMG:              AttributeDataIB =
[1695781738.671107][29270:29270] CHIP:DMG:              {
[1695781738.671126][29270:29270] CHIP:DMG:                      AttributePathIB =
[1695781738.671147][29270:29270] CHIP:DMG:                      {
[1695781738.671170][29270:29270] CHIP:DMG:                              Endpoint = 0x1,
[1695781738.671198][29270:29270] CHIP:DMG:                              Cluster = 0x28,
[1695781738.671222][29270:29270] CHIP:DMG:                              Attribute = 0x0000_0005,
[1695781738.671243][29270:29270] CHIP:DMG:                      }
[1695781738.671265][29270:29270] CHIP:DMG:
[1695781738.671294][29270:29270] CHIP:DMG:                      Data = "dfsdfasdf" (9 chars),
[1695781738.671351][29270:29270] CHIP:DMG:              },
[1695781738.671380][29270:29270] CHIP:DMG:
[1695781738.671401][29270:29270] CHIP:DMG:      ],
[1695781738.671426][29270:29270] CHIP:DMG:
[1695781738.671449][29270:29270] CHIP:DMG:      moreChunkedMessages = false,
[1695781738.671472][29270:29270] CHIP:DMG:      InteractionModelRevision = 11
[1695781738.671493][29270:29270] CHIP:DMG: },
[1695781738.671556][29270:29270] CHIP:DMG: Received group attribute write for Group=1234 Cluster=0x0000_0028 attribute=0x0000_0005
[1695781738.671852][29270:29270] CHIP:DMG: Processing group attribute write for endpoint=0 Cluster=0x0000_0028 attribute=0x0000_0005
[1695781738.671885][29270:29270] CHIP:DMG: AccessControl: checking f=1 a=g s=0xFFFFFFFFFFFF04D2 t= c=0x0000_0028 e=0 p=m
[1695781738.671917][29270:29270] CHIP:DMG: AccessControl: allowed
[1695781738.672126][29270:29270] CHIP:DL: writing settings to file (/tmp/chip_kvs-L9evKa)
[1695781738.672993][29270:29270] CHIP:DL: renamed tmp file to file (/tmp/chip_kvs)
[1695781738.673066][29270:29270] CHIP:DMG: Endpoint 0, Cluster 0x0000_0028 update version to eaa6893c

yunhanw-google, 2023-09-27:
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/Interaction-Model.adoc#924-valid-write-attribute-paths

yunhanw-google, 2023-09-27:
GroupId and endpointId should be mutual exclusive

joonhaengHeo, 2023-09-27:
The invoke message also operates even if it includes an endpointId.

yunhanw-google, 2023-09-27:
Thanks for the feedback. @joonhaengHeo We would fix it for both invoke and write in IM code.