project-chip/connectedhomeip #29470: [CADMIN-1.16] Step 13 expected outcome mismatch when executed in YAML.

manjunath-grl, 2023-09-27:
### Reproduction steps

Summary:
According to the test-plan Step 13 out come of fabric attribute list should be in order [ FabricIndex=1, FabricIndex=3, FabricIndex=4] but got [FabricIndex=1, FabricIndex=4, FabricIndex=3] when executed.

Diff link:  https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:CADMIN_1_16_Step_13_Failure

Reproduction steps:
Step 1: Add TC-CADMIN-1.16 test in ciTests.json file.
Step 2:  Regen
            `./scripts/tools/zap_regen_all.py --type tests`
Step 3:  Build:
```
scripts/run_in_build_env.sh './scripts/build_python.sh --install_virtual_env out/venv --include_yamltests'
  ./scripts/run_in_build_env.sh \
     "./scripts/build/build_examples.py \
        --target linux-x64-chip-tool-no-ble-tsan-clang \
        --target linux-x64-all-clusters-no-ble-tsan-clang \
        --target linux-x64-lock-no-ble-tsan-clang \
        --target linux-x64-ota-provider-no-ble-tsan-clang \
        --target linux-x64-ota-requestor-no-ble-tsan-clang \
        --target linux-x64-tv-app-no-ble-tsan-clang \
        --target linux-x64-bridge-no-ble-tsan-clang \
        build \"
```
Step 4: Run YAML tests: 
```
./scripts/run_in_build_env.sh \
                  "./scripts/tests/run_test_suite.py \
                     --chip-tool -with-python \
                     --chip-tool ./out/linux-x64-chip-tool-no-ble-tsan-clang/chip-tool \
                     run \
                     --iterations 1 \
                     --all-clusters-app ./out/linux-x64-all-clusters-no-ble-tsan-clang/chip-all-clusters-app \
                     --lock-app ./out/linux-x64-lock-no-ble-tsan-clang/chip-lock-app \
                     --ota-provider-app ./out/linux-x64-ota-provider-no-ble-tsan-clang/chip-ota-provider-app \
                     --ota-requestor-app ./out/linux-x64-ota-requestor-no-ble-tsan-clang/chip-ota-requestor-app \
                     --tv-app ./out/linux-x64-tv-app-no-ble-tsan-clang/chip-tv-app \
                     --bridge-app ./out/linux-x64-bridge-no-ble-tsan-clang/chip-bridge-app \"

```

### Bug prevalence

Everytime when executed

### GitHub hash of the SDK that was being used

7b4df1c4be146f172e6e12d9658a1add0d838c41

### Platform

other

### Anything else?

Test-plan screenshot:
[image]
Execution logs:
![image](https://github.com/projec
[CADMIN_1_16_Logs.txt](https://github.com/project-chip/connectedhomeip/files/12740154/CADMIN_1_16_Logs.txt)
t-chip/connectedhomeip/assets/102359958/1ba77a5c-0651-4015-bc92-8a591455a18e)

bzbarsky-apple, 2023-09-27:
Spec does not guarantee order here.  Either the test plan or the test script need to be fixed, @manjunath-grl @cjandhyala

cjandhyala, 2023-09-27:
even the test plan is not explicitly saying the order should be followed, it say , verify the list shows (has) FabricIndex=1, FabricIndex=4 and FabricIndex=3 . @manjunath-grl in the script, as long you verify those elements are present in the response, we are good.

manjunath-grl, 2023-10-04:
Closing this issue as it is understood as resolved from the above comments.