project-chip/connectedhomeip #29471: [SVE] TC-IDM-1.2 python script fail if the device support Custom Cluster.

doublemis1, 2023-09-27:
### Reproduction steps

Define custom cluster out of the Matter Specification with the cluster id e.g. 1111111,
and run TC-IDM-1.2.
The test will fail in step 3:
```
[MatterTest] 09-28 00:54:20.770 ERROR Exception occurred in test_TC_IDM_1_2.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/mobly/base_test.py", line 783, in exec_one_test
    test_method()
  File "/root/python_testing/matter_testing_support.py", line 1008, in async_runner
    return asyncio.run(body(*args, **kwargs))
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/root/python_testing/TC_IDM_1_2.py", line 127, in test_TC_IDM_1_2
    cluster = Clusters.ClusterObjects.ALL_CLUSTERS[cid]
KeyError: 1111111
```

Based on the Matter Specification, the application may create its own Cluster for their purposes.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

Matter 1.2 SVE tag

### Platform

other

### Anything else?

[TC_IDM_1_2.patch.txt](https://github.com/project-chip/connectedhomeip/files/12740922/TC_IDM_1_2.patch.txt)
Uploading possible patch to the python script, to ignore cluster ids if the cluster id cannot be mapped with a known clusters.

raju-apple, 2023-09-27:
@cecille can you please check if the patch is fine ?

cecille, 2023-09-27:
I have a PR up with a fix for this already - https://github.com/project-chip/connectedhomeip/pull/29460

closed 2023-09-27 by mergify[bot]