project-chip/connectedhomeip #29487: Deploying an app to AppStore with custom Matter.framework fails

DanijelBojcic, 2023-09-28:
### Reproduction steps

I have an issue with deploying our app to AppStore with our custom Matter.framework.
I know it's not directly related to this repository, but I'm confident I'm not the only one experiencing this issue. I came across an open [issue](https://github.com/CocoaPods/CocoaPods/issues/11874) that seems similar, so I thought sharing it here might help accelerate the progress toward a solution.

I have a slightly modified Matter, which has some custom attributes, etc. The app builds and runs successfully locally. 
After we deploy to TestFlight, we get the following review message:

> ITMS-90565: Invalid Bundle Structure - - The framework at '/Payload/app.app/Frameworks/Matter.framework' is not permitted. Do not include platform frameworks in your app.

The message was not clear, so we wrote a ticket to the Apple Code Level Support. We got this response:
> Apple’s iOS SDK contains a framework named Matter, so your framework’s name is conflicting with the one that iOS provides. Please rename your Matter framework to something that does not match one of Apple’s frameworks.

Based on the response, I have renamed almost everything in `src/darwin/Framework`, but precisely:
- renamed `Matter` target to `PierreMatter`
- renamed every `MTR` occurrence to `PRMTR`
- renamed `<Matter/` imports to `<PierreMatter/`
- renamed every `MTR*` file to `PRMTR*`
- renamed `Matter.h` to `PierreMatter.h`
- changed the bundle identifier

Basically no naming has left from the factory version of the Matter SDK, but the issue still remains. I wrote to support once again, but no answers yet.

Maybe I should mention that the matter implementation is written in a separate private Pod.
The structure looks like this:
- App
- Pods:
    - PierreMatter
        - PierreMatter.xcframework

`PierreMatter.podspec` :
```rb

Pod::Spec.new do |s|
  ...
  s.pod_target_xcconfig = {
    "DEFINES_MODULE" => "YES",
    "USE_HEADERMAP" => "YES",
    'GCC_PREPROCESSOR_DEFINITIONS' => 'CHIP_SYSTEM_CONFIG_USE_SOCKETS=1',
  }

  # All source files that should be publicly visible
  s.source_files  = [
    "ios/**/*.{m,mm,swift,h}",
  ]

  s.private_header_files = [
    "ios/PierreMMKV/*.h",
    "ios/PierreMatter/*.h",
    "PierreOffline.xcframework/**/Headers/*.h"
  ]

  s.requires_arc  = true
  
  s.vendored_frameworks = 'PierreOffline.xcframework'
end

```

However, I don't think this causes the issue, since we have other Pods that also include xcframeworks.

Does anyone have any ideas on what I should do to resolve this issue?

### Bug prevalence

.

### GitHub hash of the SDK that was being used

.

### Platform

darwin

### Platform Version(s)

.

DanijelBojcic, 2023-09-28:
Well, it turns out the issue is caused by the private pod, but I doesn't get why...

woody-apple, 2023-10-19:
Sorry about this, but we don't handle App Store policy here. If you supply the feedback ticket, we may be able to follow up, but the App Store is the right place to ask this question of.

closed 2023-10-19 by woody-apple