project-chip/connectedhomeip #29505: [Darwin] Constraints arithmetic operations build error.

manjunath-grl, 2023-09-29:
### Reproduction steps

Summary: 
[Test Suites - Darwin (no-ble-asan-clang)] was failing in PR [29339](https://github.com/project-chip/connectedhomeip/pull/29339) since according to the specification the minimum value for MaxMeasuredValue is `MinMeasuredValue + 1` and the YAML MinValue was modified as MaxMeasuredValue + 1, Darwin does not seem to support the arithmetic operations of the constraints.

Diff link: https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:Darwin_Constraints_Arthmetic_Operaton_Error

Reproduction steps:
Run TC-FLW-2.1 or TC-RH-2.1 with the patch in [Test Suites - Darwin (no-ble-asan-clang)](https://github.com/project-chip/connectedhomeip/actions/runs/6328638633/job/17187259315?pr=29339#logs)

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

433e6769389c5e127e8d88418d1dae3150e39534

### Platform

darwin

### Anything else?

Spec screenshot:
[image]
Execution logs:
[logs_2103922.zip](https://github.com/project-chip/connectedhomeip/files/12762692/logs_2103922.zip)
[image]

bzbarsky-apple, 2023-09-29:
This is going to be fixed by moving to the python driver instead of codegen tests.

woody-apple, 2023-10-20:
I believe this is now resolved?

bzbarsky-apple, 2023-10-21:
This is not resolved....  The relevant tests have just beed disabled on darwin-framework-tool.

woody-apple, 2025-01-29:
@manjunath-grl Are you able to check to see if this test is still disabled, and/or if it is passing now? Please re-open if still an issue.

closed 2025-01-29 by woody-apple