project-chip/connectedhomeip #29506: [Workflow] Update Ubuntu version to 22.04 for docker integration

deveritec-rosc, 2023-09-29:
### Workflow issues

Update the ubuntu version in `integrations/docker/images/base/chip-build` and `integrations/docker/images/base/chip-build-minimal` to 22.04.

This should get rid of errors due to outdated packages, such as #28435 etc.
It would also allow to run e.g. Zephyr twister based tooling using the container without the need to separate containers (e.g. for the nrfconnect-sdk, Zephyr is used, so this container can also be seen as kinda superset of the Zephyr container).

### Platform

all

dmadisetti, 2024-02-04:
Update also comes with the security issue of outdated OpenSSL: https://github.com/project-chip/connectedhomeip/issues/25688

deveritec-rosc, 2024-05-23:
Ubuntu 24.04 is now available.

Since I didn't set up the original Dockerfile I have no complete overview what is todo, from a quick glance one may also update gcc, clang and make to recent versions.