project-chip/connectedhomeip #29507: [Build] Passed all the checks stuck on the "Solving dependencies requirements"

BartJozwicki, 2023-09-29:
### Build issue(s)

1. source export.sh in (esp-idf 5.1 directory)
2. source scripts/activate.sh (connectedhomeip directory)
3. Then I run all the tests with ninja etc. it passes
4. cd examples/lighting-app/esp32
5. idf.py clean
6. idf.py --preview set-target esp32c6
Then my terminal get stucked on "Solving dependencies requirements" picture attached. I have reinstalled the enviroment and esp-idf 3 times I tried to set turget fo different boards with no avail. Please advise what to do. 
[image]

### Platform

esp32

bzbarsky-apple, 2023-09-29:
@shubhamdp @dhrishi

dhrishi, 2023-09-30:
@BartJozwicki In general, at this step, the `managed_components` directory is created by pulling in the required components from the Espressif component manager. Can you please wait for some time and see if it completes? Or it is stuck there for a longer time?

shubhamdp, 2023-09-30:
@BartJozwicki can you please take a look at https://github.com/project-chip/connectedhomeip/issues/27653#issuecomment-1635375210, looks like similar problem.

BartJozwicki, 2023-09-30:
> @BartJozwicki can you please take a look at [#27653 (comment)](https://github.com/project-chip/connectedhomeip/issues/27653#issuecomment-1635375210), looks like similar problem.

Removing cache helped, thank you my problem has been resolved.

closed 2023-09-30 by BartJozwicki