project-chip/connectedhomeip #29518: [BUG] [CHEF]  Can't add OTA Requestor

jvmahon, 2023-09-30:
### Reproduction steps

Is this a but - unable to add OTA requestor to a CHEF example device ...

1. Start with a simple Chef example, e.g., rootnode_onofflightswitch_FsPlMr090Q
2. build for LInux:  ./chef.py -zb -t linux -d rootnode_onofflightswitch_FsPlMr090Q
3. All is OK,
4. Go back and edit rootnode_onofflightswitch_FsPlMr090Q.zap by adding an "OTA Requestor" endpoint (0x0012)
5. Repeat step 2 to build with the additional endpoint
6. 

### Bug prevalence

Each time I try to use

### GitHub hash of the SDK that was being used

latest SDK as of Sept. 30

### Platform

other

### Platform Version(s)

Linux, maybe others

### Anything else?

After adding an OTA Requestor, build halts with the following message ...

============================

Activating environment (setting environment variables):

  Setting environment variables for CIPD package manager...done
  Setting environment variables for Project actions........skipped
  Setting environment variables for Python environment.....done
  Setting environment variables for pw packages............skipped
  Setting environment variables for Host tools.............done

Checking the environment:

20230930 11:47:09 INF Environment passes all checks!

Environment looks good, you are ready to go!

Running ZAP script to generate artifacts
Searching for zcl file from /home/jvm33/Downloads/connectedhomeip/examples/chef/devices/rootnode_onofflightswitch_FsPlMr090Q.zap
🔧 Using temporary state directory: /tmp/zapvfxcwqve/zap.VHvAdj
🤖 ZAP generation started: 
    🔍 input files: /home/jvm33/Downloads/connectedhomeip/examples/chef/devices/rootnode_onofflightswitch_FsPlMr090Q.zap
    🔍 output pattern: /home/jvm33/Downloads/connectedhomeip/examples/chef/out/rootnode_onofflightswitch_FsPlMr090Q/zap-generated/
    🔍 using templates: /home/jvm33/Downloads/connectedhomeip/src/app/zap-templates/matter-idl-server.json
    🔍 using zcl data: /home/jvm33/Downloads/connectedhomeip/src/app/zap-templates/zcl/zcl.json
    🔍 zap version: ver. 2023.9.22, featureLevel 98, commit: 62d5aae8bb0ada3e4790157b7eb11f1984fb9489 from 2023-09-22T18:27:32.000Z, mode: binary, exe: zap-cli
🕐 Setup time: 8s 257ms 
👉 using input file: /home/jvm33/Downloads/connectedhomeip/examples/chef/devices/rootnode_onofflightswitch_FsPlMr090Q.zap
👉 using output destination: /home/jvm33/Downloads/connectedhomeip/examples/chef/out/rootnode_onofflightswitch_FsPlMr090Q/zap-generated/
🕐 File loading time: 511ms
🤖 Generating files:
    ✍  /home/jvm33/Downloads/connectedhomeip/examples/chef/out/rootnode_onofflightswitch_FsPlMr090Q/zap-generated/Clusters.matter
🕐 Generation time: 1s 104ms 
Files generated in: /home/jvm33/Downloads/connectedhomeip/examples/chef/out/rootnode_onofflightswitch_FsPlMr090Q/zap-generated/
RPC PW disabled
Software Version String: ""
Product ID 0x8000 / Vendor ID 0xFFF1
Building...
aarch64
Done. Made 265 targets from 222 files in 780ms
ninja: Entering directory `out'
[1/51] ACTION //:chef-data-model_codegen_codegen(//third_party/connectedhomeip/build/toolchain/linux:linux_arm64_gcc)
FAILED: gen/app/PluginApplicationCallbacks.h gen/app/callback-stub.cpp 
python ../third_party/connectedhomeip/third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../ --current-path .. --default-toolchain=//third_party/connectedhomeip/build/toolchain/linux:linux_arm64_gcc --current-toolchain=//third_party/connectedhomeip/build/toolchain/linux:linux_arm64_gcc --capture-output --python-dep-list-files gen/chef-data-model_codegen_codegen_metadata_path_list.txt -- ../third_party/connectedhomeip/scripts/codegen.py --generator cpp-app --output-dir gen --expected-outputs gen/chef-data-model_codegen.expected.outputs ../../devices/rootnode_onofflightswitch_FsPlMr090Q.matter
2023-09-30 11:47:27 INFO    Parsing idl from ../../devices/rootnode_onofflightswitch_FsPlMr090Q.matter
Traceback (most recent call last):
  File "/home/jvm33/Downloads/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/lark/lexer.py", line 590, in lex
    yield lexer.next_token(lexer_state, parser_state)
  File "/home/jvm33/Downloads/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/lark/lexer.py", line 528, in next_token
    raise UnexpectedCharacters(lex_state.text, line_ctr.char_pos, line_ctr.line, line_ctr.column,
lark.exceptions.UnexpectedCharacters: No terminal matches 'U' in the current parser context, at line 1502 col 46

am      attribute updateState default = Unknown;
                                        ^
Expected one of: 
	* FALSE
	* TRUE
	* MINUS
	* ESCAPED_STRING
	* POSITIVE_INTEGER
	* HEX_INTEGER

Previous tokens: Token('EQUAL', '=')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/jvm33/Downloads/connectedhomeip/examples/chef/linux/out/../third_party/connectedhomeip/scripts/codegen.py", line 178, in <module>
    main(auto_envvar_prefix='CHIP')
  File "/home/jvm33/Downloads/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/jvm33/Downloads/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/jvm33/Downloads/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/jvm33/Downloads/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/jvm33/Downloads/connectedhomeip/examples/chef/linux/out/../third_party/connectedhomeip/scripts/codegen.py", line 125, in main
    idl_tree = CreateParser().parse(open(idl_path, "rt").read())
  File "/home/jvm33/Downloads/connectedhomeip/examples/chef/linux/third_party/connectedhomeip/scripts/py_matter_idl/matter_idl/matter_idl_parser.py", line 519, in parse
    idl = self.transformer.transform(self.parser.parse(file))
  File "/home/jvm33/Downloads/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/lark/lark.py", line 645, in parse
    return self.parser.parse(text, start=start, on_error=on_error)
  File "/home/jvm33/Downloads/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/lark/parser_frontends.py", line 96, in parse
    return self.parser.parse(stream, chosen_start, **kw)
  File "/home/jvm33/Downloads/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/lark/parsers/lalr_parser.py", line 41, in parse
    return self.parser.parse(lexer, start)
  File "/home/jvm33/Downloads/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/lark/parsers/lalr_parser.py", line 171, in parse
    return self.parse_from_state(parser_state)
  File "/home/jvm33/Downloads/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/lark/parsers/lalr_parser.py", line 193, in parse_from_state
    raise e
  File "/home/jvm33/Downloads/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/lark/parsers/lalr_parser.py", line 183, in parse_from_state
    for token in state.lexer.lex(state):
  File "/home/jvm33/Downloads/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/lark/lexer.py", line 599, in lex
    raise UnexpectedToken(token, e.allowed, state=parser_state, token_history=[last_token], terminals_by_name=self.root_lexer.terminals_by_name)
lark.exceptions.UnexpectedToken: Unexpected token Token('ID', 'Unknown') at line 1502, column 46.
Expected one of: 
	* FALSE
	* POSITIVE_INTEGER
	* MINUS
	* TRUE
	* ESCAPED_STRING
	* HEX_INTEGER
Previous tokens: [Token('EQUAL', '=')]

[2/51] ACTION //:chef-data-model_zapgen_zap_pregen(//third_party/connectedhomeip/build/toolchain/linux:linux_arm64_gcc)
ninja: build stopped: subcommand failed.
Traceback (most recent call last):
  File "/home/jvm33/Downloads/connectedhomeip/examples/chef/./chef.py", line 985, in <module>
    sys.exit(main())
  File "/home/jvm33/Downloads/connectedhomeip/examples/chef/./chef.py", line 846, in main
    shell.run_cmd("ninja -C out")
  File "/home/jvm33/Downloads/connectedhomeip/examples/chef/stateful_shell.py", line 139, in run_cmd
    raise RuntimeError(
RuntimeError: Error. Nonzero return code.
Returncode: 1
Cmd: ninja -C out

bzbarsky-apple, 2023-10-02:
Looks like on the new endpoint the updateState default value was set to "Unknown", but the lexer for .matter files expects an integer (which makes sense, since the ZAP stuff does not really handle named enum values as defaults; you should be using the actual numeric value as the default).

jvmahon, 2023-10-03:
As a follow-up, I tried looking at the zap file for the example "OTA-Requestor-App" (https://github.com/project-chip/connectedhomeip/tree/master/examples/ota-requestor-app/ota-requestor-common).  This had me even more confused as the .zap file didn't include the "OTA Requestor" (0x0012) endpoint and, instead, had a "Matter Secondary Network commissioning" endpoint (0xF002). Is there documentation somewhere explaining why this is the case? I couldn't find it.

closed 2023-10-30 by jvmahon