project-chip/connectedhomeip #29552: [Bug]? Cannot get a successful response to removeNetwork on Darwin platform.

maurycyw, 2023-10-03:
### Reproduction steps

When recovering from commissioning failures or service failures post commissioning, we would like to call removeNetwork followed by removeFabric but are having issues on Darwin.

Reproduction steps:
```
guard let networkCluster = MTRBaseClusterNetworkCommissioning(
    device: mtrBaseDevice, /// we also use this device to successfully read from BasicInfoCluster
    endpointID: 0,
    queue: .main
) else {
    return
}
        
let params = MTRNetworkCommissioningClusterRemoveNetworkParams()
let data = networkSSID.data(using: .utf8)! /// networkSSID is a String
params.networkID = data

do {
    try await networkCluster.removeNetwork(with: params)
} catch {
    logger.log(level: .error, "Unable to remove the network: \(networkSSID) \(error).")
}
```

always returns: `Error Domain=MTRInteractionErrorDomain Code=202 "Operation was not successful."`

Has anyone else run into issues regarding calling removeNetwork as an admin? Is the conversion `networkSSID.data(using: .utf8)! ` valid for `params.networkID` ?

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

4088a77f557e8571a39338fad51a1d8eb0131d79

### Platform

darwin

### Platform Version(s)

16.4.1

### Anything else?

We have a matching implementation for Android and have no issues there.

bzbarsky-apple, 2023-10-04:
> always returns: Error Domain=MTRInteractionErrorDomain Code=202 "Operation was not successful."

@maurycyw that's `MTRInteractionErrorCodeFailsafeRequired`, right?

The spec for RemoveNetwork says:

> If this command is received without an armed fail-safe context (see Section 11.9.6.2, “ArmFailSafe Command”), then this command SHALL fail with a FAILSAFE_REQUIRED status code sent back to the initiator.

In the code snippet above it doesn't look like a fail-safe is being armed...

What you need to is:

1. Arm a fails-safe.
2. RemoveNetwork.
3. Clear the fail-safe by sending a CommissioningComplete command (yes, the naming is a bit odd....)

> Is the conversion networkSSID.data(using: .utf8)! valid for params.networkID ?

Technically there is no standard for how SSIDs are to be interpreted as strings...  But in practice yes, UTF-8 is the only sane thing to do there.

closed 2023-10-04 by bzbarsky-apple

bzbarsky-apple, 2023-10-05:
That's sort of a Matter spec issue, not an API issue per se... But yes, we really need to figure out a better way to document APIs that are actually being automatically generated from the XML that describes the spec...