project-chip/connectedhomeip #29559: Very High CPU and Memory Utilization While Building the Chip Tool

VaishaliAvhale, 2023-10-04:
### Reproduction steps

While building the chip tool (using 'ninja -C out/debug'), due to high CPU and memory utilization, eventually led to the termination of open processes or a system restart.

I have followed the chip-tool build steps from below document:
https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/chip_tool_guide.md

System Specification:
Kernal: 6.2.0-26-generic
[image]

CPU and Memory Utilization:
[image]

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

54a146018c1b042b347b04e1ff7b075903b55603 

### Platform

core

bzbarsky-apple, 2023-10-04:
High CPU is normal; ninja is just trying to use all the compute available.

I expect your real issue is the memory.  It looks like you have 16GB of RAM available; do you have any swap configured @VaishaliAvhale ?  chip-tool linking is pretty memory-intensive...

That said, it's _possible_ that using `-j1` for `ninja` would help with the memory too.  It looks like `scripts/examples/gn_build_example.sh` (which I assume you are using if you are following the above-linked docs) does not have support for passing a `-j` argument to ninja.  But if you modify that script's last line to read `ninja -j1 -C ...`, does that help?

VaishaliAvhale, 2023-10-05:
Swap is configured on my system, and its size is 2GB. While building the chip tool, it was getting filled. I have increased its size to 4GB, and now I'm able to successfully build the chip tool in a single go.

I also tried to build a chip tool using the -j4 argument it works too but it is a bit slower.
Thanks, @bzbarsky-apple.

bzbarsky-apple, 2023-10-05:
Yeah, OK.  Needing >2GB to link is not great, but not surprising given how chip-tool is set up...

closed 2023-10-06 by VaishaliAvhale