project-chip/connectedhomeip #29590: [MW320] CONFIG_5GHz_SUPPORT seems to be tested without being defined

bzbarsky-apple, 2023-10-05:
### Reproduction steps

Enable `-Wundef`.  https://github.com/project-chip/connectedhomeip/actions/runs/6419947097/job/17430925897?pr=29582

```
INFO    ../../examples/all-clusters-app/nxp/mw320/third_party/connectedhomeip/third_party/nxp/mw320_sdk/repo/middleware/wifi/wifidriver/mlan_cfp.c:1660:5: error: "CONFIG_5GHz_SUPPORT" is not defined, evaluates to 0 [-Werror=undef]
INFO     1660 | #if CONFIG_5GHz_SUPPORT
INFO          |     ^~~~~~~~~~~~~~~~~~~
```

@doru91 

### Platform

other

### Type

CI tested

crlonxp, 2023-10-11:
A pull-request is created for this issue: https://github.com/project-chip/connectedhomeip/pull/29702
Please help to review.
Thank you

bzbarsky-apple, 2023-10-12:
@crlonxp Is this fixed now?

crlonxp, 2023-10-12:
Yes, it is.

closed 2023-10-12 by bzbarsky-apple