project-chip/connectedhomeip #29611: [BUG] [1.1] [Infineon] [PSoC6] cannot finish commissioning

mmt-10, 2023-10-06:
### Reproduction steps

PSoC6 cannot finish commissioning.
It stacked in WiFi setting process on commissioning.

When PSoC6 got this situation, it cannot be Factory Reset.
WiFi process blocked.

```bash
./chip-tool pairing ble-wifi 1234 $SSID $PASSWORD 20202021 3840
```

[infineon-lighting-app-20231004.txt](https://github.com/project-chip/connectedhomeip/files/12826817/infineon-lighting-app-20231004.txt)

> CHIP:DL: WiFi station mode change: Disabled -> Enabled
> CHIP:DMG: Decreasing reference count for CommandHandler, remaining 1
> CHIP:DL: Starting P6 WiFi layer
> CHIP:DL: EnableStationMode
> CHIP:DL: WiFi station state change: NotConnected -> Connecting
> CHIP:DL: Attempting to connect WiFi station interface
> CHIP:DL: Connecting to AP : [WIFIAP_SSID] 
> 
> CHIP:DL: Write received for CHIPoBLE RX characteristic con:8000 handle:000c len2
> CHIP:DL: CHIPoBLE unsubscribe received
> CHIP:DL: BLE GATT connection closed (con 32768, reason 19)

[chip-tool-20231004.txt](https://github.com/project-chip/connectedhomeip/files/12826816/chip-tool-20231004.txt)

> [1696563218.555536][150802:150802] CHIP:CTL: Shutting down the commissioner
> [1696563218.555556][150802:150802] CHIP:CTL: Shutting down the controller
> [1696563218.555561][150802:150802] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
> [1696563218.555649][150802:150802] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
> [1696563218.555688][150802:150802] CHIP:FP: Shutting down FabricTable
> [1696563218.555691][150802:150802] CHIP:TS: Pending Last Known Good Time: 2023-10-06T09:30:35
> [1696563218.555705][150802:150802] CHIP:TS: Previous Last Known Good Time: 2023-10-06T09:30:35
> [1696563218.555711][150802:150802] CHIP:TS: Reverted Last Known Good Time to previous value
> [1696563218.555955][150802:150802] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-zeQqRA)
> [1696563218.556216][150802:150802] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
> [1696563218.556233][150802:150802] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
> [1696563218.556237][150802:150802] CHIP:DL: Inet Layer shutdown
> [1696563218.556240][150802:150802] CHIP:DL: BLE shutdown
> [1696563218.556558][150802:150802] CHIP:DL: System Layer shutdown
> [1696563218.556879][150802:150802] CHIP:TOO: Run command failure: src/app/CommandSender.cpp:271: CHIP Error 0x00000032: Timeout

### Bug prevalence

always

### GitHub hash of the SDK that was being used

commit 8f66f4215bc0708efc8cc73bda80620e67d8955f (HEAD, tag: v1.1.0.1)

### Platform

other

bzbarsky-apple, 2023-10-06:
@ifyall not much to go on in those logs... The device is doing:
```
CHIP:DL: Connecting to AP : [WIFIAP_SSID] 
```

and then no device logs and the client eventually times out because it never got a response to EnableNetwork.

mmt-10, 2023-10-16:
How about this inquiry?
Did you be reproduce?

ifyall, 2023-10-16:
We have created a ticket in our internal backlog and will investigate it in our next sprint, starting on Monday 23 Oct.

praveenCY, 2023-10-27:
Wi-Fi commissioning issue mentioned in this is not reproducible in my setup, but factory reset button not working after issue is fixed in PR : 30043

mmt-10, 2023-10-27:
How about door-lock,?
door-lock has same WiFi issue.

Chip-Tool which commit id : 309e9f1afc989cea5927063d8a592e5eecb0b3ea was working.
However,  issue occurred with Chip-Tool of v1.1.0.1.

mmt-10, 2023-11-06:
@praveenCY 
Thank you for your reply.
How do I remove commissioning data from PSoC6?
PSoC6 has incomplete commissioning data.
And, Factory reset button does not working.

PSoC6 is stacked, because it cannot join matter fabric and cannot remove commissioning data by Factory Reset button.

praveenCY, 2023-11-06:
@mmt-10   : can you try with PR : https://github.com/project-chip/connectedhomeip/pull/30043 changes for factory reset button problem ? 

If you are still seeing the problem, please add below call in function "Init" of KeyValueStoreManagerImpl.cpp (src/platform/Infineon/PSOC6) temporarily to clear all the keys. Revert this changes after as it will causes keys to be removed everything during bootup.

[image]

closed 2023-11-27 by mergify[bot]