project-chip/connectedhomeip #29614: [CERT-TEST-FAILURE] - TC-SMOKECO - Test failed

KishokG, 2023-10-06:
### Feature Area

Area:SmokeCO2

### Test Case

TC-SMOKECO-2.5 & 2.6

### Reproduction steps

1. Bring up the DUT with all-clusters-app using this command `./chip-all-clusters-app --enable-key 00112233445566778899aabbccddeeff`
2. Pair the DUT using chip-tool through "onnetwork" transport.
3. Start the interactive mode `./chip-tool interactive start`
4. Run the below commands,
`smokecoalarm subscribe battery-alert 3 30 1 1`
`smokecoalarm subscribe interconnect-smoke-alarm 3 30 1 1`
`generaldiagnostics test-event-trigger hex:00112233445566778899aabbccddeeff 0xffffffff00000095 1 0`

### Expected Outcome:
		
> [1696574766.017984][3595:3597] CHIP:DMG: 				StatusIB =
> [1696574766.018056][3595:3597] CHIP:DMG: 				{
> [1696574766.018131][3595:3597] CHIP:DMG: 					status = 0x00 (SUCCESS),
> [1696574766.018207][3595:3597] CHIP:DMG: 				},
> 
**Report from DUT after successfull event trigger:**
> [1696574766.022820][3595:3597] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_005C Attribute 0x0000_0003 DataVersion: 2745071379
> [1696574766.022903][3595:3597] CHIP:TOO:   BatteryAlert: 1
> [1696574766.023007][3595:3597] CHIP:DMG: Refresh LivenessCheckTime for 34224 milliseconds with SubscriptionId = 0x3bff817d Peer = 01:0000000000000001

### Actual Outcome:
> [1696574766.017984][3595:3597] CHIP:DMG: 				StatusIB =
> [1696574766.018056][3595:3597] CHIP:DMG: 				{
> [1696574766.018131][3595:3597] CHIP:DMG: 					status = 0x00 (SUCCESS),
> [1696574766.018207][3595:3597] CHIP:DMG: 				},

### Bug prevalence

Whenever I try to trigger the test event after subscribing more than one attributes.

### GitHub hash of the SDK that was being used

1ec98c8d962d2847a3325e9ee1542907300f8c9b

### Platform

raspi

### Anything else?

- While running TC-SMOKECO2.5 and TC-SMOKECO-2.6 test cases in interactive mode, If we're subscribing to more than one attribute before triggering the event using the EventTrigger field key the DUT sends only the SUCCESS response. 
- In this scenario DUT should send particular attribute report to the chip-tool after the SUCCESS response, but we're not getting any response from the DUT after the success response.

ericzijian1994, 2023-10-07:
Because the second subscription stops the first, you need to add "--keepSubscriptions".

KishokG, 2023-10-09:
As per your comment I tried with `--keepSubscriptions true` and It's working fine. Thank you @ericzijian1994

closed 2023-10-09 by KishokG