project-chip/connectedhomeip #29616: [BUG] How can i sign noc bymyself to finish commissioning

freemanspace, 2023-10-06:
### Reproduction steps

i use chiptool-Android-master project to do this
i want sign noc by myself ,so i set NOCChainIssuer to do this
but don't work,
please help!

code as below=======================================

deviceController.setNOCChainIssuer { csrInfo, attestationInfo ->
   var controllerParams = ControllerParams.newBuilder()
          .setControllerVendorId(deviceInfo.vendorId)
          .setRootCertificate(byteArrayOf(rootCertificate))
          .setIntermediateCertificate(byteArrayOf(intermediateCertificate))
          .setOperationalCertificate(byteArrayOf(operationalCertificate))
          .setIpk(byteArrayOf(opk))
          .build()
    Log.d(TAG,"finish init controllerParams")
    Log.d(TAG,"start onNOCChainGeneration")
    deviceController.onNOCChainGeneration(controllerParams)   <----it will lock no log to print
    Log.d(TAG,"end onNOCChainGeneration")
}

log as below=============================================
2023-10-06 15:04:48.563  7989-8012  CTL                     com.google.chip.chiptool             D  Received certificate signing request from the device
2023-10-06 15:04:48.564  7989-8012  CTL                     com.google.chip.chiptool             D  Successfully finished commissioning step 'SendOpCertSigningRequest'
2023-10-06 15:04:48.564  7989-8012  CTL                     com.google.chip.chiptool             D  Commissioning stage next step: 'SendOpCertSigningRequest' -> 'ValidateCSR'
2023-10-06 15:04:48.564  7989-8012  CTL                     com.google.chip.chiptool             D  Performing next commissioning step 'ValidateCSR'
2023-10-06 15:04:48.618  7989-8012  CTL                     com.google.chip.chiptool             D  Successfully finished commissioning step 'ValidateCSR'
2023-10-06 15:04:48.618  7989-8012  CTL                     com.google.chip.chiptool             D  Commissioning stage next step: 'ValidateCSR' -> 'GenerateNOCChain'
2023-10-06 15:04:48.618  7989-8012  CTL                     com.google.chip.chiptool             D  Performing next commissioning step 'GenerateNOCChain'
2023-10-06 15:04:48.618  7989-8012  CTL                     com.google.chip.chiptool             D  Getting certificate chain for the device from the issuer
2023-10-06 15:04:48.618  7989-8012  CTL                     com.google.chip.chiptool             D  Parsing Certificate Signing Request
2023-10-06 15:04:48.642  7989-8012  TOO                     com.google.chip.chiptool             D  VerifyCertificateSigningRequest
2023-10-06 15:04:48.643  7989-8012  DeviceProv...ngFragment com.google.chip.chiptool             D  finish init controllerParams
2023-10-06 15:04:48.643  7989-8012  DeviceProv...ngFragment com.google.chip.chiptool             D  start onNOCChainGeneration

### Bug prevalence

every time

### GitHub hash of the SDK that was being used

#29574

### Platform

android

bzbarsky-apple, 2023-10-06:
> it will lock no log to print

Does it help if you make this call async?  Because my guess is that the Android bits are calling into your callback with the Matter stack lock held (which if true seems like a very strange thing to do), then `onNOCChainGeneration` tries to take the lock and deadlocks.

@yunhanw-google

freemanspace, 2023-12-05:
deviceController.onNOCChainGeneration(controllerParams) 
must be run on mainthread
change to main Thread will work

closed 2023-12-05 by freemanspace

bhavin-android, 2025-01-29:
Any way I can do it locally for test purpose as for now don't have server. So how to generate this locally and set ?