project-chip/connectedhomeip #29645: Cannot change wifi ssid after remove all fabric

nykh2010, 2023-10-08:
https://github.com/project-chip/connectedhomeip/blob/f34b13cd861dae644d6c08ba20440de30c151bb8/src/platform/ESP32/NetworkCommissioningDriver.cpp#L163C1

I found a commissioned device remove all fabric cannot change wifi ssid and password in new commission stage.
The detail step is below:
1. Use Smart Things to add a ESP32 lighting device. Set wifi ssid "wifi1".
2. After commission complete, delete device from Smart Things app. Then the device will reopen commissioning window.
3. Reboot device, and scan the QR code start a commission stage.
4. Change wifi ssid "wifi2", not same with first step.
5. The commssion will abort, because the wifi ssid could not be set success. 
Because after remove fabric, the wifi credentials not be erase from nvs. So in the second commission stage, device reject new wifi configuration. In this case, I think it canbe update network, instead of rejecting.

bzbarsky-apple, 2023-10-09:
> Because after remove fabric, the wifi credentials not be erase from nvs

This is a spec violation right there, no?  Per spec, removing the last fabric should remove all settings configured via Matter, including network configurations...

@dhrishi @shubhamdp

nykh2010, 2023-10-10:
> > Because after remove fabric, the wifi credentials not be erase from nvs
> 
> This is a spec violation right there, no? Per spec, removing the last fabric should remove all settings configured via Matter, including network configurations...
> 
> @dhrishi @shubhamdp

I'm not found it in spec? Should developer do this by himself?
In esp-matter repo, it also not do this process. https://github.com/espressif/esp-matter/blob/63e79f02114cc795ad9e4c68303076bb89cac925/examples/light/main/app_main.cpp#L84

dhrishi, 2023-10-14:
@nykh2010 The spec does not explicitly mention about what is to be done in such cases. I have raised a spec issue to provide clarity (private repo - https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/7635). For now, you can remove to network config in the application followed by advertisement over BLE or use the reset to factory option.

dhrishi, 2024-10-25:
Closing this now as  the resolution is provided

closed 2024-10-25 by dhrishi