project-chip/connectedhomeip #29655: [BUG]

aravindone, 2023-10-09:
### Reproduction steps

i tried to read the attribute of Node label and Description.

`E (183703) all-clusters-app: =======activetext read error=======
E (183713) all-clusters-app: Failed to read Nodelabel attribute. Status: 0xC3
I (183723) TAG: zclString: 
I (183723) chip[ZCL]: 0x3f409cb8ep 1 clus 0x0000_0028 attr 0x0000_0005 not supported
E (183733) all-clusters-app: Failed to set Nodelabel attribute. Status: 0xC3
I (183733) TAG: zclString: 
E (183743) all-clusters-app: Failed to read Description attribute. Status: 0x8D
I (183753) TAG: zclString: ` 

im using 
`namespace ElectricalMeasurement {
namespace Attributes {

namespace MeasurementType {
EmberAfStatus Get(chip::EndpointId endpoint, uint32_t * value); // bitmap32
EmberAfStatus Set(chip::EndpointId endpoint, uint32_t value);
} // namespace MeasurementType

// etc`
from `zzz_generated/app-common/app-common/zap-generated/attributes/Accessors.h`
Get function is not working for fetching the data .

### Bug prevalence

everytime

### GitHub hash of the SDK that was being used

a3c4b48365

### Platform

esp32, core

VaishaliAvhale, 2023-10-09:
Used latest connectedhomeip commit id: f34b13cd861dae644d6c08ba20440de30c151bb8 
Flashed and commissioned the all-cluster-app and read NodeLabel and Description attribute both are supported on my end. please check it in the attached log file.
[all-cluster-app.log](https://github.com/project-chip/connectedhomeip/files/12845962/all-cluster-app.log)
[all-cluster-app-esp32.log](https://github.com/project-chip/connectedhomeip/files/12845972/all-cluster-app-esp32.log)

Please attach complete DUT and chip-tool logs and also tell me the connectedhomeip's commit ID.

bzbarsky-apple, 2023-10-09:
@aravindone I don't know how you are using `ElectricalMeasurement::Attributes::MeasurementType` anything to get "node label and description".  Please be more specific about what exactly you are doing.

aravindone, 2023-10-10:
Hi both sorry for the confusion 
@VaishaliAvhale i need to read the data from the esp 32 itself using the get set functions in the `zzz_generated/app-common/app-common/zap-generated/attributes/Accessors.h`

@bzbarsky-apple i have modified the question, now  i need to read the data using Get Set function

bzbarsky-apple, 2023-10-11:
@aravindone Assuming that the error logs above correspond to the provided code:
```
Failed to read Nodelabel attribute. Status: 0xC3
```
That's UNSUPPORTED_CLUSTER.  Do you have that cluster enabled on endpoint 0?

```
 (183743) all-clusters-app: Failed to read Description attribute. Status: 0x8D 
```
That's INVALID_DATA_TYPE.  That's because you are trying to read into a 0-size buffer, so what were you expecting to happen, exactly?

The right way to read strings is more like this:
```
   char value_chars[N]; //  N needs to be at least the max size of the string data as defined in the spec
   chip::MutableCharSpan value(value_chars);
   status = ModeSelect::Attributes::Description::Get(DeviceID, value); // NOTE: Will modify `value.size()`

   // NOTE: value_chars is NOT NULL-TERMINATED, so doing things like
   //
   // ESP_LOGI(TAG, "Retrieved value: %s", value.data());
   //
   // is a great way to have security bugs.  Instead use:
   //
   // ESP_LOGI(TAG, "Retrieved value: %.*s", static_cast<int>(value.size()), value_.data());

   // Now if you want to use the same buffer for a different read, you need a new MutableCharSpan.
```

closed 2023-10-11 by bzbarsky-apple

aravindone, 2023-10-16:
Hi @bzbarsky-apple thanks for the reply.