project-chip/connectedhomeip #29656: [BUG]

tkaewplik, 2023-10-09:
### Reproduction steps

I have tried compiling the latest tag (SVE_23_09/rc1) because I would like to pair switch bot hub 2 with Matter hub (RP4).

but I got an error while compiling about 
```
requires feature level 98, we only have 97. Please upgrade your zap!
```
1. Is this something that I have to downgrade zap?

the full log can be seen in the attachment below
[pi4_compile.log](https://github.com/project-chip/connectedhomeip/files/12844831/pi4_pair.log)

**Note**
If you are curious about why don't I use the older version instead, I can successfully compile tag v1.1.0.1 on RP4 but I have a timeout problem after I paired the devices together so I think using a new version might solve it (I used to compile SVE_23_09/rc1 tag on Linux Amd 64bits, and It can pair and read measured data from Switch bot hub 2)

pairing Matter hub (v1.1.0.1) with switch bot hub 2 
[pi4_pairing_v1_1.log](https://github.com/project-chip/connectedhomeip/files/12845122/pi4_pairing_v1_1.log)
using parts-list and get a timeout error from v1.1.0.1 on RP4
[pi4_parts_list_v1_1.log](https://github.com/project-chip/connectedhomeip/files/12844918/pi4_parts_list.log)

2. Is the timeout problem from the attachment above a bug on v1.1.0.1 (RP4), Is there anything I can investigate to solve this timeout error?
3. Actually, the measured data I got from the Switch bot hub 2 when I tried with Linux Amd usually be 0, is this unstable thing normal in Matter?

### Bug prevalence

When ever I do

### GitHub hash of the SDK that was being used

5ccb774e7e5fe7ead3ff85bec1cda31a459c9c37

### Platform

raspi

### Platform Version(s)

SVE_23_09/rc1

bzbarsky-apple, 2023-10-09:
@tkaewplik It sounds like you have a ZAP that claims feature level 97 set up, but the revision you are trying to compile needs a ZAP that claims feature level 98.

Where exactly is your ZAP installed?  Are you using the ZAP that's pulled in by the in-tree bootstrap, or something else?

bzbarsky-apple, 2023-10-09:
That part is fine, as expected.  But what ZAP are you actually _using_ for your build?  Do you have any `ZAP_*` environment variables defined that would override which ZAP gets used (e.g. `ZAP_INSTALL_PATH` or `ZAP_DEVELOPMENT_PATH`)?

tkaewplik, 2023-10-11:
Thank you for pointing out the environment vars, I found the path to another zap directory already.

```
ubuntu@ubuntu:~$ echo $ZAP_INSTALL_PATH

ubuntu@ubuntu:~$ echo $ZAP_DEVELOPMENT_PATH
/home/ubuntu/zap
```

however, after I pulled the new code and rebuilt the folder (npm run build), I still could not compile the connewctedhomeip project., I am stuck on building chip-tool-utils.DiscoverCommand.cpp forever.

```
ubuntu@ubuntu:~/connectedhomeip$ ninja -C out/host
ninja: Entering directory `out/host'
[0/776] c++ obj/examples/chip-tool/commands/discover/chip-tool-utils.DiscoverCommand.cpp.o
```

can I ignore this example folder on Ninja? will it cause any problems?

bzbarsky-apple, 2023-10-11:
> however, after I pulled the new code and rebuilt the folder (npm run build)

Or you could unset the env var and use the ZAP from bootstrap, of course.  Not least because then you will be getting the right version....

> I am stuck on building chip-tool-utils.DiscoverCommand.cpp forever

That is very odd.  Are you sure you are stuck on that and not on one of the unit tests that are just not being logged by the build system?

> will it cause any problems?

Do you plan to use chip-tool?  If not, no.

tkaewplik, 2023-10-12:
> That is very odd. Are you sure you are stuck on that and not on one of the unit tests that are just not being logged by the build system?
I have waited for more than 8 hours and It still shows the same message below so have no other idea just to say It is about that file. do I have to check something else?
```
ubuntu@ubuntu:~/connectedhomeip$ ninja -C out/host
ninja: Entering directory `out/host'
[0/776] c++ obj/examples/chip-tool/commands/discover/chip-tool-utils.DiscoverCommand.cpp.o
```

> Do you plan to use chip-tool? If not, no.
In case I would like to use Python, I can ignore it right? 
My apologies for asking this, could you tell me how to by pass it?

One more question, I have tried pairing Matter hub (v1.1.0.1) with switch bot hub 2 as I mentioned in the issue body. however, It has errors while running parts list command ([pi4_parts_list.log](https://github.com/project-chip/connectedhomeip/files/12877148/pi4_parts_list.log)) , so can you confirm me something, is this a bug which need an updated version?
```
[1696846854.166922][40520:40522] CHIP:EM: Failed to Send CHIP MessageCounter:240084191 on exchange 36801i sendCount: 4 max retries: 4
[1696846856.636718][40520:40522] CHIP:SC: CASESession timed out while waiting for a response from the peer. Current state was 1
[1696846856.636816][40520:40522] CHIP:IN: SecureSession[0xffff7c00adb0]: Released - Type:2 LSID:30326
[1696846856.636913][40520:40522] CHIP:DIS: OperationalSessionSetup[1:0000000000000001]: State change 4 --> 2
[1696846856.636989][40520:40522] CHIP:DIS: OperationalSessionSetup[1:0000000000000001]: State change 2 --> 1
[1696846856.637063][40520:40522] CHIP:-: src/protocols/secure_channel/CASESession.cpp:485: CHIP Error 0x00000032: Timeout at examples/chip-tool/commands/clusters/ModelCommand.cpp:62
```

maybe I could go back to v1.1.0.1 and try fixing that again.

bzbarsky-apple, 2023-10-12:
> do I have to check something else?

Well, you could check what processes are actually running.  Is anything taking CPU?  Is there an actual "c++" process running?  Is there any process running whose name starts with "Test"?

>  My apologies for asking this, could you tell me how to by pass it?

Don't do the "build everything" build, which is what you are doing right now...

> is this a bug which need an updated version?

That just looks like packets are not getting delivered to the discovered IP (or maybe back from it to the sender of the Sigma1 message?).  I would check whether that IP can actually be pinged.

tkaewplik, 2023-10-12:
> That just looks like packets are not getting delivered to the discovered IP (or maybe back from it to the sender of the Sigma1 message?). I would check whether that IP can actually be pinged.

I returned to v1.1.0.1 and unexpectedly found that It could read the data from RP4 without any changes from last time, maybe there was a network problem in the past.

> Well, you could check what processes are actually running. Is anything taking CPU? Is there an actual "c++" process running? Is there any process running whose name starts with "Test"?

I haven't checked it yet, but my second terminal can not respond to a thing while it compiling that file (could be about the network or CPU anyway), I will stick with v1.0.1 for now.

I really appreciate your support!

closed 2023-10-12 by tkaewplik