project-chip/connectedhomeip #29659: [Tooling] Python 3.12 support

emanuelbuholzer, 2023-10-09:
### Reproduction steps / Feature

Setting up the Python environment using `bootstrap.sh` fails when using python3.12. This  originates from the backwards incompatible change removing the deprecated `imp` module. For more information about the `imp` module see: https://discuss.python.org/t/how-do-i-migrate-from-imp/27885

```
[459/1012] python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../.. --default-toolchain=//build/toolchain/linux:linux_x64_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --capture-output --module piptools --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files python/gen/matter_build_venv._compile_requirements_metadata_path_list.txt -- compile --resolver=backtracking --allow-unsafe --output-file python/gen/matter_build_venv/compiled_requirements.txt python/gen/matter_build_venv/generated_requirements.txt ../../scripts/setup/requirements.build.txt
FAILED: python/gen/matter_build_venv/compiled_requirements.txt 
python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../.. --default-toolchain=//build/toolchain/linux:linux_x64_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --capture-output --module piptools --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files python/gen/matter_build_venv._compile_requirements_metadata_path_list.txt -- compile --resolver=backtracking --allow-unsafe --output-file python/gen/matter_build_venv/compiled_requirements.txt python/gen/matter_build_venv/generated_requirements.txt ../../scripts/setup/requirements.build.txt
    error: subprocess-exited-with-error
    
    × Getting requirements to build wheel did not run successfully.
    │ exit code: 1
    ╰─> [54 lines of output]
        running egg_info
        writing lib/PyYAML.egg-info/PKG-INFO
        writing dependency_links to lib/PyYAML.egg-info/dependency_links.txt
        writing top-level names to lib/PyYAML.egg-info/top_level.txt
        Traceback (most recent call last):
          File "/home/emanuelbuholzer/connectedhomeip/.environment/gn_out/python-venv/lib64/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
            main()
          File "/home/emanuelbuholzer/connectedhomeip/.environment/gn_out/python-venv/lib64/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
            json_out['return_val'] = hook(**hook_input['kwargs'])
                                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
          File "/home/emanuelbuholzer/connectedhomeip/.environment/gn_out/python-venv/lib64/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
            return hook(config_settings)
                   ^^^^^^^^^^^^^^^^^^^^^
          File "/tmp/pip-build-env-xslmrga5/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 355, in get_requires_for_build_wheel
            return self._get_build_requires(config_settings, requirements=['wheel'])
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
          File "/tmp/pip-build-env-xslmrga5/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 325, in _get_build_requires
            self.run_setup()
          File "/tmp/pip-build-env-xslmrga5/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 341, in run_setup
            exec(code, locals())
          File "<string>", line 288, in <module>
          File "/tmp/pip-build-env-xslmrga5/overlay/lib/python3.12/site-packages/setuptools/__init__.py", line 103, in setup
            return distutils.core.setup(**attrs)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
          File "/tmp/pip-build-env-xslmrga5/overlay/lib/python3.12/site-packages/setuptools/_distutils/core.py", line 185, in setup
            return run_commands(dist)
                   ^^^^^^^^^^^^^^^^^^
          File "/tmp/pip-build-env-xslmrga5/overlay/lib/python3.12/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
            dist.run_commands()
          File "/tmp/pip-build-env-xslmrga5/overlay/lib/python3.12/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
            self.run_command(cmd)
          File "/tmp/pip-build-env-xslmrga5/overlay/lib/python3.12/site-packages/setuptools/dist.py", line 989, in run_command
            super().run_command(command)
          File "/tmp/pip-build-env-xslmrga5/overlay/lib/python3.12/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
            cmd_obj.run()
          File "/tmp/pip-build-env-xslmrga5/overlay/lib/python3.12/site-packages/setuptools/command/egg_info.py", line 318, in run
            self.find_sources()
          File "/tmp/pip-build-env-xslmrga5/overlay/lib/python3.12/site-packages/setuptools/command/egg_info.py", line 326, in find_sources
            mm.run()
          File "/tmp/pip-build-env-xslmrga5/overlay/lib/python3.12/site-packages/setuptools/command/egg_info.py", line 548, in run
            self.add_defaults()
          File "/tmp/pip-build-env-xslmrga5/overlay/lib/python3.12/site-packages/setuptools/command/egg_info.py", line 586, in add_defaults
            sdist.add_defaults(self)
          File "/tmp/pip-build-env-xslmrga5/overlay/lib/python3.12/site-packages/setuptools/command/sdist.py", line 113, in add_defaults
            super().add_defaults()
          File "/tmp/pip-build-env-xslmrga5/overlay/lib/python3.12/site-packages/setuptools/_distutils/command/sdist.py", line 251, in add_defaults
            self._add_defaults_ext()
          File "/tmp/pip-build-env-xslmrga5/overlay/lib/python3.12/site-packages/setuptools/_distutils/command/sdist.py", line 336, in _add_defaults_ext
            self.filelist.extend(build_ext.get_source_files())
                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
          File "<string>", line 204, in get_source_files
          File "/tmp/pip-build-env-xslmrga5/overlay/lib/python3.12/site-packages/setuptools/_distutils/cmd.py", line 107, in __getattr__
            raise AttributeError(attr)
        AttributeError: cython_sources
        [end of output]
    
    note: This error originates from a subprocess, and is likely not a problem with pip.
[... 159 lines cut ...]
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

### Platform

other

### Platform Version(s)

1ec98c8d96

### Type

Manually tested with SDK

### (Optional) If manually tested please explain why this is only manually tested

Found in manual workflow after verifying the Fedora 39 release candidate.

mraichev, 2024-01-23:
The same issue has been replicated on trying to install Notificore Python Library https://github.com/Notificore/notificore-python.git

Processing /Users/maxraychev/Dev/practice/python/notificore/notificore-python
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [24 lines of output]
      Traceback (most recent call last):
        File "/Users/maxraychev/Dev/practice/python/notificore/my_env/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/Users/maxraychev/Dev/practice/python/notificore/my_env/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/Users/maxraychev/Dev/practice/python/notificore/my_env/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/pz/hsbprgsn1nl05d08mg0ksnjw0000gp/T/pip-build-env-pz8sfodi/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 325, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=['wheel'])
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/pz/hsbprgsn1nl05d08mg0ksnjw0000gp/T/pip-build-env-pz8sfodi/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 295, in _get_build_requires
          self.run_setup()
        File "/private/var/folders/pz/hsbprgsn1nl05d08mg0ksnjw0000gp/T/pip-build-env-pz8sfodi/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 480, in run_setup
          super(_BuildMetaLegacyBackend, self).run_setup(setup_script=setup_script)
        File "/private/var/folders/pz/hsbprgsn1nl05d08mg0ksnjw0000gp/T/pip-build-env-pz8sfodi/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 311, in run_setup
          exec(code, locals())
        File "<string>", line 2, in <module>
        File "/Users/maxraychev/Dev/practice/python/notificore/notificore-python/notificore_restapi/__init__.py", line 3, in <module>
          from .api import Requester, Recipient, Response, Price, APIError
        File "/Users/maxraychev/Dev/practice/python/notificore/notificore-python/notificore_restapi/api.py", line 9, in <module>
          import requests
      ModuleNotFoundError: No module named 'requests'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip

rosahay-silabs, 2024-03-18:
faced similar on a new mac setup by installing `brew install python3` used `brew install python@3.11` and it did work. I guess its working on an older mac since the package might be compiled and cached.

andy31415, 2024-10-21:
We require python 3.10 or above, so 3.9 has a very high chance of failure (at least in the future).

In the mean time, our mac builds were updated to use python 3.11 as bundled (and not try to use system python) where as linux compiles with python 3.12. I expect this bug to actually be fixed now since we moved to ubuntu 24 and python 3.12 there.

closed 2024-10-21 by andy31415