project-chip/connectedhomeip #29696: Have callbacks from the SessionManager/TransportMgr to notify Exchanges of connection closures

pidarped, 2023-10-10:
### Feature description

This option would allow the SessionManager to appropriately notify each open exchange that the underlying connection is broken/closed. This should allow the application to take necessary actions on the Exchange. The session should also be marked as defunct. 

### Platform

core

bzbarsky-apple, 2023-10-11:
> The session should also be marked as defunct.

Defunct, or evicted?  When sessions are evicted, their exchanges get notified automatically.  At least for secure sessions, which are the only type that have a concept of "defunct" anyway.  It seems to me that if our TCP connection closed, the session should just be evicted, since we are not going to be _receiving_ messages on them, and the only difference between "defunct" and "evicted" in practice is what happens when we suddenly get a message from the other side on the session.

For unauthenticated sessions over TCP (aka CASE/PASE handshakes), we're going to need to figure out what to do with connection closure.

pidarped, 2023-10-11:
For TCP based sessions, I was thinking that, in order to align with retaining the session for a long term, we could mark them as defunct when connection is lost, and make them active again when the connection is re-established. So, slightly different than how the session becomes active for MRP. The reason I did not want to evict it is because, IIUC, the persistent state of the session is also thrown away, rt? So, then, for the new connection, the session would have to be established from scratch, instead of session resumption.

bzbarsky-apple, 2023-10-12:
> and make them active again when the connection is re-established

Ah, that makes sense.  In that case, yes, you need some other mechanism to handle the exchanges.  Are exchanges tied to a specific connection, unlike sessions?

PR #30339 (2024-05-01): Connection and CASE session management over TCP
Currently, connecting to a peer is coupled with sending a message to the peer.
This decouples the two and creates a clear API for connecting to a peer address. Goes along with the existing Disconnect() API. This would be essential during activation of retained sessions by solely connecting to the peer and associating with the retained session.

Surface Connection completion and Closure callbacks and hook them through SessionManager(TransportMgr delegate) and CASESession.

Modify ActiveConnectionState in TCPBase to hold state for each connection,  [... cut]
closed 2024-05-01 by pidarped