project-chip/connectedhomeip #29713: After implementing multi-admin, how should I use CHIP-TOOL to query all administrators of the matter device?

whqxbs, 2023-10-11:
### Test issue(s)

After implementing multi-admin, how should I use CHIP-TOOL to query all administrators of the matter device?
I noticed that I can see all the administrators of the device in full in Homekit, but I didn't find a way to implement it in CHIP-TOOL, can you tell me how to go about it?

### Platform

other, all

whqxbs, 2023-10-11:
[image]

bzbarsky-apple, 2023-10-12:
```
chip-tool operationalcredentials read fabrics ${NODE_ID} 0 --fabric-filtered false
```

That is, read the "Fabrics" attribute from the "Operational Credentials" cluster on endpoint 0 of the node, and turn off fabric-filtering, so you get all the entries for all the fabrics.

closed 2023-10-12 by bzbarsky-apple

whqxbs, 2023-10-12:
I have tried the commond you say, but I can only get the information below. 
I still can't get the admins' name like what Homekit can do in the picture above.
[image]

whqxbs, 2023-10-12:
> ```
> chip-tool operationalcredentials read fabrics ${NODE_ID} 0 --fabric-filtered false
> ```
> 
> That is, read the "Fabrics" attribute from the "Operational Credentials" cluster on endpoint 0 of the node, and turn off fabric-filtering, so you get all the entries for all the fabrics.

bzbarsky-apple, 2023-10-12:
> I have tried the commond you say, but I can only get the information below.

Please attach (as a text attachment, not paste, not an image) the full log from the command involved, including the chip-tool command line.

Because either you left off `--fabric-filtered false` or this is not the same device as the "Connected Services" screen shot above (which seems to show three fabrics, whereas here the new screenshot only shows one).

bzbarsky-apple, 2023-10-12:
But also: I am pretty sure the strings shown above do not come from the device.  "Matter Test" just means the VendorId was 0xFFF1 == 65521.  "Apple Home" means the vendor ID was 0x1349, etc.

whqxbs, 2023-10-12:
Thank you for your reply! I corrected the command I entered and can successfully see 3 fabrics with CHIP-TOOL connected to the device at the same time as Homekit (image and text are below).
I would also like to ask why three fabrics and what is this extra VendorID: 4996?
[result.txt](https://github.com/project-chip/connectedhomeip/files/12878629/result.txt)
[image]

whqxbs, 2023-10-12:
> 4996 == 0x1384 is "Apple Keychain". That's the fabric that lets you have that "Connected Services" UI even if a device is removed from Apple Home, lets you easily commission the device onto other fabrics even if it gets removed from Apple Home, etc.

Could you tell me how to implement what you said about continuing to implement that commission the device onto other fabrics after deleting the device in Homekit's Home? I would like to know the relevant operations and commands, thank you!

bzbarsky-apple, 2023-10-12:
See https://developer.apple.com/documentation/mattersupport