project-chip/connectedhomeip #29729: [BUG]Facing Issue While Executing YAML TestCases

VaishaliAvhale, 2023-10-12:
### Reproduction steps

I have followed the below README file to execute the YAML test cases. However, the 'tests' command has been removed from 'connectedhomeip' and it is no longer supported. 
https://github.com/project-chip/connectedhomeip/blob/master/src/app/tests/suites/README.md

I've tried out the other ways to execute the YAML test cases using the following command, but I have encountered an issue while running the test cases.

```
$ ./scripts/tests/run_test_suite.py --runner chip_tool_python --target 'Test_TC_OO_1_1' --log-level debug run --iterations 1 --test-timeout-seconds 120
```

Can someone please assist me with this?

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

ecc0d63cf7eb91f4017bf8c264b53cf690420eb5

### Platform

core

### Platform Version(s)

v1.2

bzbarsky-apple, 2023-10-12:
> but I have encountered an issue while running the test cases.

What issue?  This is not actionable as filed...  Please attach logs from the failure.

cecille, 2023-10-13:
The problem here is the tests command has been removed, but the readme has not been updated and the runner documentation is either non-existent or at least hard to find. I have not yet been able to find it.

To run tests locally now, you need to fire up an interactive chip session, then use runner.py or one of the other wrappers to run the yaml. 

@vivien-apple would you be able to provide instructions on which wrapper to use and how to pass in the required flags?

bzbarsky-apple, 2023-10-13:
Is the issue "how to run the test against some device that is not all-clusters-app"?

bzbarsky-apple, 2023-10-13:
OK, if I hack up the harness scripts a bit and run them with `--dry-run`, I get this "what did I run" output, paths modified to not be absolute, and the chip-tool paths are pointing to my chip-tool, and of course the --storage-directory is whatever temp folder the harness created.  
```
python3 scripts/tests/yaml/chiptool.py pairing code 0x12344321 MT:-24J042C00KA0648G00 --server_path ./out/debug/chip-tool-tests/chip-tool --server_arguments 'interactive server' --storage-directory /var/folders/ty/gbwtrbq52rsg6r6jn35dcp9c0000gn/T/tmppgjhfl1v

python3 scripts/tests/yaml/chiptool.py tests Test_TC_OO_1_1 --PICS src/app/tests/suites/certification/ci-pics-values --server_path ./out/debug/chip-tool-tests/chip-tool --server_arguments 'interactive server' --storage-directory /var/folders/ty/gbwtrbq52rsg6r6jn35dcp9c0000gn/T/tmppgjhfl1v
```

Running that second command on its own after manually commissioning things seems to work right.

bzbarsky-apple, 2023-10-13:
And yes, we desperately need to update the documentation....

bzbarsky-apple, 2023-10-13:
https://github.com/project-chip/connectedhomeip/pull/29752 to improve the dry-run logging so it gives the info I pasted above.

cecille, 2023-10-17:
Assigning this over to Vivien as he has the best handle on how to run this and can provide documentation.

cecille, 2023-10-17:
FYI - most of the command flags are NOT documented properly in the help menu, but flow through to the lower layers, so you can use more flags than are exposed by the help menus.