project-chip/connectedhomeip #29738: [BUG] Thread commissioning fails when using the Linux lighting example with the Thread feature

MonicaisHer, 2023-10-12:
### Reproduction steps

### Set Up:
#### Version
- nRF Connect SDK version: 2.4.2
- Matter SDK version: 1.1.0.1
- Thread version: 1.3
- RCP API version: 6

#### Machine A:
- [Linux lighting app](https://github.com/project-chip/connectedhomeip/tree/v1.1.0.1/examples/lighting-app/linux)
- [OTBR(OpenThread Border Router) agent](https://github.com/openthread/ot-br-posix/tree/thread-reference-20230119)
- [RCP](https://github.com/openthread/ot-nrf528xx) on nRF52840 dongle
- OS: Ubuntu Classic 23.04 for amd64

Linux lighting app has been executed with the Thread flag using the following commands:
```
cd ~/connectedhomeip/examples/lighting-app/linux
gn gen out/debug
ninja -C out/debug
cd out/debug
./chip-lighting-app --thread
```

#### Machine B:
- chip-tool
- OTBR agent
- RCP on nRF52840 dongle
- OS: Ubuntu Core 22.04 for arm64

A Thread network has been successfully formed by OTBR.

### Logs during Thread commissioning:
Once both machines are set up, I used chip-tool to attempt to commission the lighting app into the Thread network. However, the Thread commissioning failed with the following log:
- lighting-app (Machine A): [full logs](https://pastebin.ubuntu.com/p/xsvDsKJ6Fr/)
- OTBR agent (Machine A):
```
openthread-border-router.otbr-agent[65506]: otbr-agent[65506]: [ERR ]-UTILS---: Replied to org.freedesktop.DBus.Properties.GetAll with result io.openthread.Error.NotFound
otbr-agent[65506]: 00:02:39.273 [W] Platform------: Error processing result: NotFound
otbr-agent[65506]: 00:02:39.273 [W] Platform------: Error waiting response: NotFound
otbr-agent[65506]: 00:02:39.273 [W] Platform------: Get Coex Metrics failed: NotFound
otbr-agent[65506]: [ERR ]-UTILS---: Replied to org.freedesktop.DBus.Properties.GetAll with result io.openthread.Error.NotFound
```

- chip-tool (Machine B): [full logs](https://pastebin.ubuntu.com/p/SMbTP5RysS/)
```
CHIP:CTL: Required network information not provided in commissioning parameters
CHIP:CTL: Parameters supplied: wifi (no) thread (yes)
CHIP:CTL: Device supports: wifi (yes) thread(no)
```
From the logs, it appears that the `--thread` flag enables the Thread feature, but the lighting app log shows that it is disabled. I guess there might be a communication failure between the lighting app and OTBR dbus daemon.

### Access the OTBR dbus server
During my investigation, I also followed these test [scripts](https://github.com/openthread/ot-br-posix/blob/main/tests/dbus/test-server) to assess the OTBR dbus server on Machine A:
```
sudo dbus-send --system --dest=io.openthread.BorderRouter.wpan0 --print-reply \
        /io/openthread/BorderRouter/wpan0 \
        io.openthread.BorderRouter.GetProperties \
        "array:string:""DeviceRole"
string "disabled"

sudo dbus-send --system --dest=io.openthread.BorderRouter.wpan0 --print-reply \
        /io/openthread/BorderRouter/wpan0 \
        io.openthread.BorderRouter.GetProperties \
        "array:string:""ThreadVersion"
uint16 4

sudo dbus-send --system --dest=io.openthread.BorderRouter.wpan0 --print-reply \
        /io/openthread/BorderRouter/wpan0 \
        io.openthread.BorderRouter.GetProperties \
        "array:string:""Uptime"
uint64 672853

$ sudo dbus-send --system --dest=io.openthread.BorderRouter.wpan0 \
        --type=method_call --print-reply /io/openthread/BorderRouter/wpan0 \
        org.freedesktop.DBus.Properties.GetAll \
        string:io.openthread.BorderRouter
Error io.openthread.Error.NotFound: io.openthread.Error.NotFound
```

### Bug prevalence

10/10

### GitHub hash of the SDK that was being used

8f66f4215bc0708efc8cc73bda80620e67d8955f/v1.1.0.1

### Platform

nrf, raspi, other

### Anything else?

There is a similar issue related to thread commissioning and OTBR dbus: https://github.com/project-chip/connectedhomeip/issues/27262

bzbarsky-apple, 2023-10-12:
> From the logs, it appears that the Thread feature has not been enabled.

Correct.  If you look at main.cpp for lighting-app on the SHA linked above (https://github.com/project-chip/connectedhomeip/commit/8f66f4215bc0708efc8cc73bda80620e67d8955f), you see that the network commissioning piece looks like this:

```
#if CHIP_DEVICE_CONFIG_ENABLE_WPA
namespace {
DeviceLayer::NetworkCommissioning::LinuxWiFiDriver sLinuxWiFiDriver;
Clusters::NetworkCommissioning::Instance sWiFiNetworkCommissioningInstance(0, &sLinuxWiFiDriver);
} // namespace
#endif
```

There is no mention of Thread, so that application on that sha does not support Thread.

That was fixed in #27861 back in July.

closed 2023-10-12 by bzbarsky-apple

MonicaisHer, 2023-10-13:
@bzbarsky-apple Thank you for your reply. I tried Thread commissioning with the lighting app built from the latest master branch, and Thread commissioning was successful.