project-chip/connectedhomeip #29760: [Test Failed] Smart bulb cannot complete pairing with CHIP-TOOL

whqxbs, 2023-10-13:
### Test issue(s)

I purchased a matter-based smart bulb and tested it to complete pairing properly with Homekit and the accompanying app, but I was unable to test pairing with CHIP-TOOL.
Prior to this, I was able to successfully share and pair with CHIP-TOOL and Homekit at the same time using matter.js.
Below is the output graph of my commands, I first paired successfully using the proprietary app, then shared the pairing code, and then after that, when I paired using CHIP-TOOL, the following error was reported.
Thanks for the answer!
[error.txt](https://github.com/project-chip/connectedhomeip/files/12895727/error.txt)

bzbarsky-apple, 2023-10-13:
> [1697214761.980234][14209:14211] CHIP:CTL: Failed in verifying 'Attestation Information' command received from the device: err 101. Look at AttestationVerificationResult enum to understand the errors

That's your error.  Error 101 is "unknown PAA".  In your command line you did not tell chip-tool to use any non-test PAAs, and presumably this device is signed by an actual non-test PAA.

Doing:
```
chip-tool pairing code 1013 16056623887 --paa-trust-store-path credentials/development/paa-root-certs
```
or 
```
chip-tool pairing code 1013 16056623887 --paa-trust-store-path credentials/production/paa-root-certs
```

to pick up the PAAs that are known about in-tree might work better.  Though if it's a new PAA that was added since the last DCL sync that might not work either.

whqxbs, 2023-10-14:
I got it. With your help, the bulb and CHIPTOOL can be paired properly. Thank you very, very much!

closed 2023-10-24 by woody-apple