project-chip/connectedhomeip #29768: [Platform] Matter.framework should set the timezone during comissioning

woody-apple, 2023-10-13:
### Reproduction steps

Currently it does not appear that we configure timezone during commissioning.

### Platform

darwin

### Type

Manually tested with SDK

woody-apple, 2023-10-13:
Let's see if we can knock this out quickly

woody-apple, 2023-10-13:
Need to figure out DST as well

bzbarsky-apple, 2023-10-14:
Impl notes: First, to avoid use-after-free, we need to fix `CommissioningParameters::ClearExternalBufferDependentValues` to clear `mTimeZone`, `mDSTOffsets`, `mDefaultNTP`.  Then we need to fix `AutoCommissioner::SetCommissioningParameters` to handle those members correctly, but in terms of the `haveMaybeDanglingBufferPointers` check and the copy bits below that.  This might want to be a separate PR.

After that, in `commissionNodeWithID:commissioningParams:error:` on MTRDeviceController, we want to:

1. Grab `auto * tz = [NSTimeZone localTimeZone]`.  
2. Create a `Clusters::TimeSynchronization::Structs::TimeZoneStruct::Type`
3. Set its validAt to 0
4. Set its offset to `tz.secondsFromGMT - tz.daylightSavingTimeOffset`
5. Set its name to `tz.name` 
6. Create a DataModel::List with that one item, and pass it to `params.SetTimeZone()`.

As far as DST.... Looks like the best we can do is specify the current and next several offsets, as follows:

7. Get `auto * nextTranstion = tz.nextDaylightSavingTimeTransition`.  If this returns nil, no DST information is needed.
8. Otherwise, create an N-element array of `Clusters::TimeSynchronization::Structs::DSTOffsetStruct::Type` (however many we want to specify)
9. Set first array element to have `offset == tz.daylightSavingTimeOffset`, `validStarting == 0`, `validUntil` set to the `nextTransition` NSDate converted into a Matter epoch-us value.
10. Set second array element to have `offset = [tz daylightSavingTimeOffsetForDate:nextOffset]`, `validStarting` same as previous validUntil, validUntil either null if this is last entry or `[tz nextDaylightSavingTimeTransitionAfterDate:nextTranstion], and so on.

Need to decide how many elements to set here, which will partly depend on how many elements `CommissioningParameters` actually accepts.  Either way it will get clamped to `DSTOffsetListMaxSize`, which is defaulted to 2 in the SDK, so probably not much point doing more than current and next.  That gives at least 6 months, and someone upstream from our framework will need to update DST as time passes anyway...

bzbarsky-apple, 2023-10-24:
> First, to avoid use-after-free

Mostly fixed in https://github.com/project-chip/connectedhomeip/pull/29908, except for mDefaultNTP, but we're not planning to do anything with that anyway.

closed 2023-10-24 by mergify[bot]