project-chip/connectedhomeip #29771: [BUG] Media cluster conformance issues on all-clusters-app

cecille, 2023-10-14:
### Reproduction steps

visible in zap file and attributes. 

Summary:
Channel - 0x504
- attribute 0x0 is included, but requires the CL feature, which is not on
- commands 0x2 and 0x3 are not included, but are mandatory

Media playback - 0x506
- includes attributes 0x1, 0x2, 0x4, 0x5, 0x6 attributes are on, but require the AS feature, which is not on
- commands 0x0, 0x01, 0x02, 0x0A are all mandatory, and none are included.

Media input - 0x507
- command 0x3 is included, but requires the NU feature, which is not on

Content launcher - 0x502
- attribute 0x0 is included, but requires the CS feature, which is not on
- attribute 0x1 is included, but requires the UP feature, which is not on

Audio output - 0x50B
- cmd 0x0 is not included, but is mandatory

Application launcher - 0x50C
- attribute 0x0 is included, but requires the AP feature, which is not on
- commands 0x00, 0x01, 0x02, 0x03 are all mandatory, but not included

Application Basic - 0x50D
- attribute 0x4 is mandatory, but is not included

Account login - 0x50e
- no commands are included, but they are all mandatory

### Bug prevalence

always

### GitHub hash of the SDK that was being used

a7c96f088bcada4b085d0580605bf34b9ab53ed5

### Platform

other

chrisdecenzo, 2023-10-16:
Here is what I recommend, so that we can build out the test cases for these clusters using all-clusters app.

Channel - 0x504
- enable all features (value=3)
- enable all attributes (0-2)
- enable all server commands (0,2,3)

Media playback - 0x506
- enable all features (value=3)
- enable all attributes (0-6)
- enable all commands (0,1,2,3,4,5,6,7,8,9,11)

Media input - 0x507
- enable all features (value=1)
- enable all attributes (0-1)
- enable all commands (0,1,2)

Content launcher - 0x502
- enable all features (value=3)
- enable all attributes (0-1)
- enable all server commands (0,1)

Audio output - 0x50B
- enable all features (value=1)
- enable all attributes (0-1)
- enable all server commands (0,1)

Application launcher - 0x50C
- enable all features (value=1)
- enable all attributes (0-1)
- enable all server commands (0,1,2)

Application Basic - 0x50D
- no features (value=0)
- enable all attributes (0-7)
- no server commands

Account login - 0x50e
- no features (value=0)
- no attributes
- enable all server commands (0,2,3)

Target Navigator - 0x505
- no features (value=0)
- enable all attributes (0-1)
- enable all server commands (0)

cecille, 2023-10-16:
Confirmed with QA that they are using the TV app for their testing, as is the CI (https://github.com/project-chip/connectedhomeip/blob/master/scripts/tests/chiptest/__init__.py#L221). Instead, it makes more sense to just disable these clusters in all-clusters as they are not used, all-clusters is already quite big.

PR #29801 (2023-10-17): Remove media clusters from all-clusters-app
Both QA and the CI are using the TV app for all testing of the media clusters. The all-clusters versions are not well tested and have conformance issues. Removing these in all-clusters in favour of the TV app.

Fixes: #29771
closed 2023-10-17 by mergify[bot]