project-chip/connectedhomeip #29774: [Darwin] Question of scanNetwork status in commissioning

joonhaengHeo, 2023-10-15:
### Feature description

Unlike other platforms, darwin seems to unconditionally skip scanNetwork during commissioning.

In the case of Android, it is supported with the flag below.
https://github.com/project-chip/connectedhomeip/blob/master/src/controller/java/src/chip/devicecontroller/ControllerParams.java#L14
https://github.com/project-chip/connectedhomeip/blob/master/src/controller/java/src/chip/devicecontroller/ControllerParams.java#L15

Is there any way to do scanNetwork?
Otherwise, I would like to inquire whether there are plans to support this feature.

### Platform

darwin

### Platform Version(s)

master

bzbarsky-apple, 2023-10-16:
There are no specific plans I am aware of, but specific suggestions for how it should work are welcome.

joonhaengHeo, 2023-10-16:
@bzbarsky-apple 
As you probably know, the scanNetwork step can retrieve SSID or Thread information from the controllee device.
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/service_device_management/CommissioningFlows.adoc (in Step 12)

I think similar to Android, in the MTRCommissioningParameters class
I think it would be a good idea to add properties for thread and Wi-Fi scan and implement a callback function in the MTRDeviceControllerDelegate class.

bzbarsky-apple, 2023-10-17:
@joonhaengHeo And then what?  There's no way, given the current setup for this stuff in the core SDK, to communicate the resulting network credentials to the place where they need to be, as far as I can tell.

Basically, the way ScanNetworks was added to the AutoComissioner is pretty broken and requires a separate AutoCommissioner to be created in addition to the one inside the CHIPDeviceController to manage things, etc.

bzbarsky-apple, 2023-10-17:
And of course anyone that wants to can just ScanNetworks before starting commissioning over the PASE session, with current Darwin framework APIs.

joonhaengHeo, 2023-10-17:
According to my analysis, if the relevant code already exists in AutoCommander and the value below is changed to True (Transport is BLE), the ScanNetwork command will send it, is there something I think wrong?

Also, is there any particular reason why ScanNetwork should not be added to AutoCommander?

bzbarsky-apple, 2023-10-17:
> the ScanNetwork command will send it

That part is fine. It's being able to continue commissioning after getting the scan results that's not fine.

> is there any particular reason why ScanNetwork should not be added to AutoCommissioner?

I don't understand the question....  It has been added there, so "should not be added" makes no sense.

joonhaengHeo, 2023-10-18:
I think I gave you some confusion.
I just asked you because of your answer that you need to implement a seperate AutoCommissioner.

bzbarsky-apple, 2023-10-18:
> that you need to implement a seperate AutoCommissioner.

We do not plan to do that, basically.

joonhaengHeo, 2023-10-18:
I understand. Thank you for your answer.

closed 2023-10-18 by joonhaengHeo