project-chip/connectedhomeip #29784: [Test Failed] Ninja test failed

BartJozwicki, 2023-10-16:
### Test issue(s)

After succesfuly building and running matter on a few devices I came back to use it again but command:
ninja -C out/host failed: [1/158] c++ obj/src/credentials/examples/libCredentials.DeviceAttestationCredsExample.cpp.o

I have attached a screenshot of the terminal, the message I received:

FAILED: obj/src/credentials/examples/libCredentials.DeviceAttestationCredsExample.cpp.o 
g++ -MMD -MF obj/src/credentials/examples/libCredentials.DeviceAttestationCredsExample.cpp.o.d -Wconversion -target x86_64-apple-macos11.0 -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wconversion -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -fmacro-prefix-map=../../= -fobjc-arc -Wconversion -std=gnu++17 -frtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -DOPENSSL_NO_ASM=1 -I../../src/include -I../../src -Igen/include -I../../zzz_generated/app-common -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/include -I../../third_party/boringssl/repo/src/include -c ../../src/credentials/examples/DeviceAttestationCredsExample.cpp -o obj/src/credentials/examples/libCredentials.DeviceAttestationCredsExample.cpp.o
../../src/credentials/examples/DeviceAttestationCredsExample.cpp:58:1: error: unknown type name 'd'
d
^
../../src/credentials/examples/DeviceAttestationCredsExample.cpp:59:11: error: expected ';' after top level declarator
CHIP_ERROR ExampleDACProvider::GetProductAttestationIntermediateCert(MutableByteSpan & out_pai_buffer)
          ^
          ;
2 errors generated.

[image]
 
I am not sure why it was working fine and suddenly started to fail.

bzbarsky-apple, 2023-10-16:
@BartJozwicki What does line 58 of `src/credentials/examples/DeviceAttestationCredsExample.cpp` look like for you?  Because it sounds like you accidentally added some stray edit to the file....

bzbarsky-apple, 2023-10-16:
For example, what does `git status src/credentials/examples/DeviceAttestationCredsExample.cpp` say?