project-chip/connectedhomeip #29811: [BUG] Unable to checkout silabs/gecko_sdk

BartJozwicki, 2023-10-17:
### Reproduction steps

As in the title, it is not a problem for me as I am using a different platform but I was wondering if you are aware of that problem. That is the output from my terminal : 

after using 

1. git pull
2. git submodule update --init

fatal: Unable to checkout '1228a95262ee099a21c6be4d35224479c8e4dde2' in submodule path 'third_party/silabs/gecko_sdk'

### Bug prevalence

Always on 2 machines

### GitHub hash of the SDK that was being used

Latest

### Platform

other

### Platform Version(s)

1.3

### Anything else?

[image]

jmartinez-silabs, 2023-10-17:
Must have been a git server error or something. The gsdk and specific commit are available here
https://github.com/SiliconLabs/gecko_sdk/commit/1228a95262ee099a21c6be4d35224479c8e4dde2

And I was able to check it out/update submodule

closed 2023-11-13 by andy31415

jamesmhaley, 2024-12-26:
It's not a bug, but it is an issue.

Gecko sdk is several GiB, so needs to be cloned with git-lfs setup.

For anyone stumbling across this issue too, I had the same problem, and needed to install `git-lfs` - this is what I did on my mac:
```
brew install git-lfs
git lfs install
# verify all is working and installed
git lfs --version
```

After this I re-ran my command and it worked. 

If needed you can cd into the folder where gecko_sdk needs to live  `connectedhomeip/third_party/silabs` and `git clone https://github.com/SiliconLabs/gecko_sdk.git`. 

Don't forget to run `git submodule update --init --recursive` after to make sure there is nothing else that got stuck.