project-chip/connectedhomeip #29813: [Test Failed] A test problem with a gateway interconnecting with a light bulb

whqxbs, 2023-10-17:
### Test issue(s)

I'm trying to test a usage scenario, I first paired and connected a matter gateway using the Tuya APP, then added a smart bulb as a gateway sub-device in the APP, then I connected both the bulb and the gateway to CHIP-TOOL. i used CHIP-TOOL to look up the bulb's fabrics and I thought that the bulb's CHIP-TOOL was only for the gateway and CHIP-TOOL, but I additionally saw that the bulb's CHIP-TOOL was only for the gateway and CHIP-TOOL, but I additionally saw that the gateway and CHIP-TOOL were the same. I used CHIP-TOOL to look up the fabrics for the bulb and I thought the only CHIP-TOOL for the bulb was the gateway and CHIP-TOOL, but I additionally saw the Tuya APP in the fabrics list. I would like to ask the reason here.

Also I have a question. How can I use CHIP-TOOL to query the sub-device details under a particular gateway? I tried multiple commands and didn't achieve results.

Thanks a lot for the answer!

whqxbs, 2023-10-17:
I'll add another question, which is to assume that user A binds two smart bulbs through gateway X, after which A is unbound from gateway X and B is bound to gateway B. After that, does it mean that B has acquired the management privileges of the two smart bulbs?

bzbarsky-apple, 2023-10-17:
When you say "gateway", do you mean a bridge?

If so, you should be able to read the PartsList attribute of the Descriptor cluster on the Aggregator endpoint of the brdge to get the list or bridged devices.

whqxbs, 2023-10-17:
> When you say "gateway", do you mean a bridge?
> 
> If so, you should be able to read the PartsList attribute of the Descriptor cluster on the Aggregator endpoint of the bridge to get the list or bridged devices.

Right I meant a bridge. i tried the method you mentioned and the PartsList comes back as follows, I can only see a list of quantities, not the specific names of the devices underneath the bridged. What should I do if I want to see more specific information about the devices under the gateway?
`
[1697558687.847038][4464:4466] CHIP:TOO:   PartsList: 2 entries
[1697558687.847046][4464:4466] CHIP:TOO:     [1]: 1
[1697558687.847051][4464:4466] CHIP:TOO:     [2]: 2
`

bzbarsky-apple, 2023-10-17:
OK, so there are three endpoints here: 0, 1, 2.

You can read the DeviceTypeList attribute from the Descriptor cluster on all three endpoints to find what device types they implement.  The bridged-device endpoints would also have a Bridged Device Basic Information cluster, which might have ProductName, etc attributes.

whqxbs, 2023-10-17:
> OK, so there are three endpoints here: 0, 1, 2.
>
> You can read the DeviceTypeList attribute from the Descriptor cluster on all three endpoints to find what device types they implement. The [...]

I used DeviceTypeList attribute from the Descriptor cluster to query the Devicetype of EP0 as 22 (0x0016) and I did not find the corresponding device type from the documentation. Is 0x0016 the Devicetype of the bulb please?

`[1697563954.932089][4921:4923] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_001D Attribute 0x0000_0000 DataVersion: 1214939692
[1697563954.932127][4921:4923] CHIP:TOO:   DeviceTypeList: 1 entries
[1697563954.932138][4921:4923] CHIP:TOO:     [1]: {
[1697563954.932148][4921:4923] CHIP:TOO:       DeviceType: 22
[1697563954.932153][4921:4923] CHIP:TOO:       Revision: 1
[1697563954.932156][4921:4923] CHIP:TOO:      }
`

whqxbs, 2023-10-17:
I get the DeviceType 22, 14, 20 for endpoints 0, 1, 2. And I have two other questions：

First：
I'm trying to test a usage scenario, I first paired and connected a matter bridge using the Tuya APP, then added a smart bulb as a bridge sub-device in the APP, then I connected both the bulb and the bridge to CHIP-TOOL. I used CHIP-TOOL to look up the bulb's fabrics and I thought that the bulb's fabrics was only CHIP-TOOL and the bridge, but I additionally saw the Tuya APP in the fabrics list. I would like to ask the reason here.
Why is that?

Second：
User A binds two smart bulbs through gateway X, after which A is unbound from gateway X and B is bound to gateway B. After that, does it mean that B has acquired the management privileges of the two smart bulbs?

bzbarsky-apple, 2023-10-18:
> I get the DeviceType 22, 14, 20

OK, so those are Root Node, Aggregator, and OTA Provider respectively. 

> I would like to ask the reason here.

Why the Tuya app added a fabric too?  That's a question for Tuya.

> After that, does it mean that B has acquired the management privileges of the two smart bulbs?

I have no idea what your setup is (and which entities A and B are: users, "gateways", or light bulbs), but since nothing in what you described assigns any privileges the answer is probably "no".

If you want actually useful help you will need to be a lot more clear about the _exact_ operations being carrried out.  Otherwise this is sort of a waste of both your time and mine.

closed 2023-10-24 by woody-apple