project-chip/connectedhomeip #29818: [Feature][Zap] Support for NameSpaces+SemanticTags from XML -> ZAP -> API

jmeg-sfy, 2023-10-17:
### Feature description

Matter 1.2 added some DeviceTypes using Descriptor Cluster with SemanticTags/NameSpaces

Specification added a new NameSpaces booklet especially for those.
These SemanticTags are hard defined and should be available as Constant properly added into the SDK's Codebase

Could be something like ->  MyNameSpace::MyCluster::kMySemTag available directly from ZAP

 We got several categories:
- CSA Common NameSpaces/SemTag
- Cluster NameSpaces/SemTag
- Manufacturer NameSpaces/SenTag
 
Today there is now way to create those properly via ZAP + XML as far as I know.
The refrigerator example hardcoded those constants into its main.cpp without clean inclusion.

We could expect new Devices to use Tags more extensively in the future.

 

### Platform

all

jmeg-sfy, 2023-10-17:
not sure about the right triage for this issue/request

bzbarsky-apple, 2023-10-17:
@jmeg-sfy This probably needs to get filed in https://github.com/project-chip/zap/issues