project-chip/connectedhomeip #29837: [BUG] how to reset all by chip-tool commands in controller

jaggerwoo, 2023-10-18:
### Reproduction steps

I don't know the correct reset steps. 
Here are my reset steps：
1. Device unsubscription use this command
``` json
subscriptions shutdown-all-for-node <node_id>
```
2. Device remove use this command
``` json
pairing unpair <node_id>
```
3. clear storage use this command
``` json
storage clear-all
```
4. reset otctl
``` json
ot-ctl factoryreset
```
5. restart server
``` json
~/connectedhomeip/aaa/chip-tool interactive server
```

My Question is:
1. Is this the correct reset steps？
2. when i use `storage clear-all` command, it can't pass parameters; i have three files in storage tmp, when i execute the command, it just clear default `chip_tool_config.ini` and `chip_tool_kvs`, but not clear `chip_tool_config.alpha.ini`, how can i clear `chip_tool_config.alpha.ini`?
```json
root@pi48g:~/connectedhomeip/aaa/chip-tool-tmp# ls
chip_tool_config.alpha.ini  chip_tool_config.ini  chip_tool_kvs
```

### Bug prevalence

nil

### GitHub hash of the SDK that was being used

nil

### Platform

raspi

bzbarsky-apple, 2023-10-18:
@jaggerwoo If you want an easy way to just reset all the local chip-tool state, make sure to run it with `--storage-directory` pointing to a directory to your choice, then remove everything in that directory to reset.

jaggerwoo, 2023-10-19:
@bzbarsky-apple thank you~

closed 2023-10-19 by jaggerwoo