project-chip/connectedhomeip #29842: [Build] Can't build successfully in the esp32 which is in the all-clusters-app

tianrongqin, 2023-10-18:
### Build issue(s)

1.$ cd path/to/esp-idf
    $ source export.sh
2.$ cd path/to/connectedhomeip
    $ source scripts/activate.sh
3.$ export IDF_CCACHE_ENABLE=1
4.$ cd examples/<app-name>/esp32
5.$ idf.py --preview set-target esp32h2
6.$ rm sdkconfig
    $ idf.py -D 'SDKCONFIG_DEFAULTS=sdkconfig_m5stack.defaults' build
6.idf.py build,  but the result is not correct.I can't build successfully.

### Platform

vscode

### Anything else?

FAILED: esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-configure /home/tianrq/connectedhomeip/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-configure 
cd /home/tianrq/connectedhomeip/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip && /usr/bin/gn --root=/home/tianrq/connectedhomeip/connectedhomeip/config/esp32 gen --check --fail-on-unused-args /home/tianrq/connectedhomeip/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip && /usr/bin/cmake -E touch /home/tianrq/connectedhomeip/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-configure
ERROR at //third_party/connectedhomeip/third_party/jsoncpp/BUILD.gn:26:1: Source file not found.
source_set("jsoncpp") {
^----------------------
The target:
  //third_party/connectedhomeip/third_party/jsoncpp:jsoncpp
has a source file:
  //third_party/connectedhomeip/third_party/jsoncpp/repo/include/json/allocator.h
which was not found.
___________________
ERROR at //third_party/connectedhomeip/third_party/jsoncpp/BUILD.gn:26:1: Source file not found.
source_set("jsoncpp") {
^----------------------
The target:
  //third_party/connectedhomeip/third_party/jsoncpp:jsoncpp
has a source file:
  //third_party/connectedhomeip/third_party/jsoncpp/repo/include/json/assertions.h
which was not found.
___________________
ERROR at //third_party/connectedhomeip/third_party/jsoncpp/BUILD.gn:26:1: Source file not found.
source_set("jsoncpp") {
^----------------------
The target:
  //third_party/connectedhomeip/third_party/jsoncpp:jsoncpp
has a source file:
  //third_party/connectedhomeip/third_party/jsoncpp/repo/include/json/config.h
which was not found.
___________________
ERROR at //third_party/connectedhomeip/third_party/jsoncpp/BUILD.gn:26:1: Source file not found.
source_set("jsoncpp") {
^----------------------
The target:
  //third_party/connectedhomeip/third_party/jsoncpp:jsoncpp
has a source file:
  //third_party/connectedhomeip/third_party/jsoncpp/repo/include/json/forwards.h
which was not found.
___________________
ERROR at //third_party/connectedhomeip/third_party/jsoncpp/BUILD.gn:26:1: Source file not found.
source_set("jsoncpp") {
^----------------------
The target:
  //third_party/connectedhomeip/third_party/jsoncpp:jsoncpp
has a source file:
  //third_party/connectedhomeip/third_party/jsoncpp/repo/include/json/json.h
which was not found.
___________________
ERROR at //third_party/connectedhomeip/third_party/jsoncpp/BUILD.gn:26:1: Source file not found.
source_set("jsoncpp") {
^----------------------
The target:
[... 224 lines cut ...]
The target:
  //third_party/connectedhomeip/third_party/nlunit-test:nlunit-test
has a source file:
  //third_party/connectedhomeip/third_party/nlunit-test/repo/src/nlunit-test.h
which was not found.
___________________
ERROR at //third_party/connectedhomeip/third_party/nlunit-test/BUILD.gn:30:1: Source file not found.
static_library("nlunit-test") {
^------------------------------
The target:
  //third_party/connectedhomeip/third_party/nlunit-test:nlunit-test
has a source file:
  //third_party/connectedhomeip/third_party/nlunit-test/repo/src/nlunit-test.c
which was not found.
[905/1388] Generating ../../gen/app-codegen/cpp-app/ap...h, ../../gen/app-codegen/cpp-app/app/callback-stub.cpp
2023-10-18 16:00:50 INFO    Parsing idl from /home/tianrq/connectedhomeip/connectedhomeip/examples/all-clusters-app/all-clusters-common/all-clusters-app.matter
2023-10-18 16:00:50 INFO    Running code generator cpp-app
2023-10-18 16:00:50 INFO    File to be generated: app/PluginApplicationCallbacks.h
2023-10-18 16:00:50 INFO    Writing new data to: /home/tianrq/connectedhomeip/connectedhomeip/examples/all-clusters-app/esp32/build/gen/app-codegen/cpp-app/app/PluginApplicationCallbacks.h
2023-10-18 16:00:50 INFO    File to be generated: app/callback-stub.cpp
2023-10-18 16:00:51 INFO    Writing new data to: /home/tianrq/connectedhomeip/connectedhomeip/examples/all-clusters-app/esp32/build/gen/app-codegen/cpp-app/app/callback-stub.cpp
2023-10-18 16:00:51 INFO    Done
ninja: build stopped: subcommand failed.

 *  终端进程“/usr/bin/bash '-c', 'ninja '”已终止，退出代码: 1。

tianrongqin, 2023-10-18:
Can yo tell me the detailed step how to realize it?I have been trying for a long time,but I still can't achieve it.The above errors persists.

bzbarsky-apple, 2023-10-19:
> I did what you said

Which exact command did you run?

Does the file `third_party/jsoncpp/BUILD.gn` exist in your tree after running that command?

tianrongqin, 2023-10-19:
I realize it now.Thank you.

closed 2023-10-19 by tianrongqin

Tim4497, 2023-11-15:
> I realize it now.Thank you.

How?

 I get:

`ninja: Entering directory [...]/matter-poc/esp/esp-matter/connectedhomeip/connectedhomeip/out/host'
ninja: error: '../../third_party/jsoncpp/repo/src/lib_json/json_reader.cpp', needed by 'obj/third_party/jsoncpp/repo/src/lib_json/jsoncpp.json_reader.cpp.o', missing and no known rule to make it`

After bootstrapping Matter, and then the same error message you provided in this ticket on `idf.py build`

> Does the file third_party/jsoncpp/BUILD.gn exist in your tree after running that command?

Yes this one exists.

Tim4497, 2023-11-15:
Okay I might have used wrong settings in VS Code, after that the `git submodule update --init`

Kartikgupta1502, 2024-08-14:
I am still facing the same issues. How were you able to solve thus?

Tim4497, 2024-08-14:
Hey, I think I just installed it from scratch again. But I do not work with matter anymore, sorry!

Kartikgupta1502, 2024-08-14:
Can you share the chiptool sdk?