project-chip/connectedhomeip #29847: LevelControl, Command: MoveToLevelWithOnOff - Not working for bridged device endpoints (eg. bridged dimmable light)

twieskot, 2023-10-18:
https://github.com/project-chip/connectedhomeip/blob/68068cbe663add7c7be8b2ff6c016e9a03b287fb/src/app/clusters/level-control/level-control.cpp#L818

For bridged dynamic endpoints (eg. a bridged dimmable light) the values of:

state->minLevel, 
state->maxLevel 

are both zero(0) here - because they were never initialized.

**_Reason:_** 
The only initialization of these values is done in:
emberAfLevelControlClusterServerInitCallback(..) which is only called for the static endpoint (dummy) containing all supported clusters.

Bridged Device Scenario
Details:  https://github.com/project-chip/connectedhomeip/blob/master/examples/bridge-app/linux/README.md)

This function is NOT called for dynamically added bridged devices during the runtime.

Therefore both values must be manually read in level-control.cpp, line 818 with:

    (void)Attributes::MinLevel::Get(endpoint, &state->minLevel);
    (void)Attributes::MaxLevel::Get(endpoint, &state->maxLevel);
    
Otherwise the Command: MoveToLevel wil not work probably

bzbarsky-apple, 2023-10-19:
> which is only called for the static endpoint (dummy) containing all supported clusters.

Sounds like the other cluster instances were not registered properly on the bridge (i.e. with the startup function pointer as part of the registration).

Note that the default "dynamic cluster" macro will just set the startup function pointer to null; if you need to register clusters that need a startup function you will need to unroll the macro and do it by hand....

Yes, this is a pain and we need a better setup.  But that's the current setup.

twieskot, 2023-10-19:
Thank you very much for your quick reply.

If I understood you correctly, it is the macro "DECLARE_DYNAMIC_CLUSTER" that doesn't initialize the cluster-function list correctly:  

`struct EmberAfCluster { ... const EmberAfGenericClusterFunction * functions; ... }`

Therefore, the cluster doesn't contain functions for startup/exit, function-types:
CLUSTER_MASK_INIT_FUNCTION / CLUSTER_MASK_SHUTDOWN_FUNCTION

These functions are called in attribute-storage.cpp, Functions: initializeEndpoint(..) / shutdownEndpoint(..) when adding or removing an endpoint during runtime.

Adding the function-pointers manually for the declared level-control cluster (struct EmberAfCluster) works out fine:

```
    const EmberAfGenericClusterFunction levelControlClusterFuncs[] = {
            (EmberAfGenericClusterFunction) emberAfLevelControlClusterServerInitCallback,
            (EmberAfGenericClusterFunction) MatterLevelControlClusterServerShutdownCallback,
    };
    EmberAfCluster levelControlCluster = bridgedDimmableLightClusters[1]; // index(1) of the level-control cluster in our code
    levelControlCluster.functions = levelControlClusterFuncs;

```

The problem with this is, that the functions (emberAfLevelControlClusterServerInitCallback, MatterLevelControlClusterServerShutdownCallback) are not exported in the header file (level-control.h). So i had to patch the header-file manually doing so.

It would be desirable, if this whole circumstance is explained within the bridge-example app. 
Eg. by adding a device containing a level-control cluster within the sample code (or with any other cluster including a init/exit routine).

As a developer it is not possible to operate the Matter stack correctly without further debugging knowledge.
Or is there another place where this behavior is documented in detail?

bzbarsky-apple, 2023-10-21:
@twieskot No, the documentation around this is very much lacking....  And yes, it would be good if the example bridge app were more useful.

twieskot, 2023-10-21:
Thanks for your support - your advice was very helpful.
Well, now there is a short documentation in this ticket :)

HARRYhelenH, 2023-11-02:
@twieskot @bzbarsky-apple 
I'm stuck at the same point. I have called the function emberAfLevelControlClusterServerInitCallback(..) but the movetoLevel command is not called when I change the level from app. I using bridge-app/esp32 sample example.
Is there anything that has to be changed to achieve the level changes.

twieskot, 2023-11-02:
The bridge-app/esp32 example just supports OnOffLights.

For a dimmable light you have to add the Cluster definitions for LevelControl and for the device (eg. dimmable light, color temperature light or extended color light), here is a short example to support a DimmableLight device:

```
// Declare LevelControl cluster attributes
DECLARE_DYNAMIC_ATTRIBUTE_LIST_BEGIN(levelControlAttrs)
    DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::ClusterRevision::Id,    INT16U, 2, 0),
    DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::FeatureMap::Id,       BITMAP32, 4, 0),
    DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::CurrentLevel::Id,        INT8U, 1, ZAP_ATTRIBUTE_MASK(WRITABLE)),
    DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::OnLevel::Id,             INT8U, 1, ZAP_ATTRIBUTE_MASK(WRITABLE)),
    DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::Options::Id,           BITMAP8, 1, 0),
    DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::MinLevel::Id,            INT8U, 1, 0),
    DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::MaxLevel::Id,            INT8U, 1, 0),
    DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::StartUpCurrentLevel::Id, INT8U, 1, 0),
    DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::RemainingTime::Id,      INT16U, 2, 0),
DECLARE_DYNAMIC_ATTRIBUTE_LIST_END();

constexpr CommandId levelControlIncomingCommands[] = {
    app::Clusters::LevelControl::Commands::MoveToLevel::Id,
    app::Clusters::LevelControl::Commands::Move::Id,
    app::Clusters::LevelControl::Commands::Step::Id,
    app::Clusters::LevelControl::Commands::Stop::Id,
    app::Clusters::LevelControl::Commands::MoveToLevelWithOnOff::Id,
    app::Clusters::LevelControl::Commands::MoveWithOnOff::Id,
    app::Clusters::LevelControl::Commands::StepWithOnOff::Id,
    app::Clusters::LevelControl::Commands::StopWithOnOff::Id,
    kInvalidCommandId,
};

DECLARE_DYNAMIC_CLUSTER_LIST_BEGIN(bridgedDimmableLightClusters)
    DECLARE_DYNAMIC_CLUSTER(OnOff::Id, onOffAttrs, onOffIncomingCommands, nullptr),
    DECLARE_DYNAMIC_CLUSTER(LevelControl::Id, levelControlAttrs, levelControlIncomingCommands, nullptr),
    DECLARE_DYNAMIC_CLUSTER(Descriptor::Id, descriptorAttrs, nullptr, nullptr),
    DECLARE_DYNAMIC_CLUSTER(BridgedDeviceBasicInformation::Id, bridgedDeviceBasicAttrs, nullptr, nullptr)
DECLARE_DYNAMIC_CLUSTER_LIST_END;

DECLARE_DYNAMIC_ENDPOINT(epDimmableLight, bridgedDimmableLightClusters);

```

Afterwards make sure to call the emberAfLevelControlClusterServerInitCallback(..) after adding the new device with:
emberAfSetDynamicEndpoint(..). With the same endpoint id, of course.

HARRYhelenH, 2023-11-03:
Thanks for the reply!
Tried the same way but level callbacks are not initialised and called.
I have called emberAfLevelControlClusterServerInitCallback() function in AddDeviceEndpoint function.

>  dev->SetEndpointId(gCurrentEndpointId);
                ret =
                **emberAfSetDynamicEndpoint(index, gCurrentEndpointId, ep, dataVersionStorage, deviceTypeList, parentEndpointId);
                emberAfLevelControlClusterServerInitCallback(dev->GetEndpointId());**
                uint8_t ed=dev->GetEndpointId();