project-chip/connectedhomeip #29850: [Documentation]

kdurlu, 2023-10-18:
### Documentation issues

I want to implement and build my own version of controller (chip-tool or chip-repl, which stands more suitable) but I couldn't find any documentation about it.

Guides in the repository describes how to build the example 'chip-tool' or pre-implemented chip-repl but It does not contain any documentation about using the library or APIs as I couldn't see any.

I only want to learn more about the generic API and how to build only the controller. I think this is a missing part in documentations. Thanks.

kdurlu, 2024-03-17:
still need help...

rlaybourn, 2024-06-21:
I am in a similar situation , cannot find any example of a controller other than the chip-tool which is a bit of a complicated beast to start on. Any simple examples of a controller and using it to read and write attributes would be very useful