project-chip/connectedhomeip #29854: [BUG] production [iOS] apps cannot commission devices with Matter Framework

maurycyw, 2023-10-18:
### Reproduction steps

Same repro steps and results as described in https://github.com/project-chip/connectedhomeip/issues/23397.

Reproduction steps (taken from ^)
```
1.pair device from qrcode (via `MTRDeviceController#commissionNode`)
2.BLE:Error writing Characteristics in Chip service on the device: [The specified UUID is not allowed for this operation.]
3.failure
```

The specific bug/complaint I am reporting is that the docs do not clearly state that the Matter Support framework is required for commissioning in production. Even the [Onboarding a Matter Device](https://developer.apple.com/documentation/matter/onboarding_a_matter_device) page uses the Matter framework in the example code. 

This is problematic since engineers can potentially waste time implementing commissioning using the Matter framework only to find out later that they must use the Matter Support framework.

I might be alone in this frustration / lack of clarity in the documentation but am posting this bug in an effort to also help others as well. 

Note: If I am completely off regarding my understanding I apologize and would like some clarity around which method for commissioning would be supported in production darwin applications.

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

5a21d17fd2bb0a48f4a356937f4741f6c9dc7975

### Platform

darwin

bzbarsky-apple, 2023-10-19:
> Even the [Onboarding a Matter Device](https://developer.apple.com/documentation/matter/onboarding_a_matter_device) page uses the Matter framework in the example code.

That page says, second paragraph of the Overview section:

> The MatterSupport framework provides an onboarding payload that you use to discover the device. To obtain this onboarding payload, implement [commissionDevice(in:onboardingPayload:commissioningID:)](https://developer.apple.com/documentation/mattersupport/matteradddeviceextensionrequesthandler/commissiondevice(in:onboardingpayload:commissioningid:)?language=objc). For more information, see [Adding Matter support to your ecosystem](https://developer.apple.com/documentation/mattersupport/adding-matter-support-to-your-ecosystem?language=objc).

and links to the MatterSupport bits involved.  You do those bits, then you proceed with the commissioning via the Matter framework once you have the data needed for that.

@maurycyw What could we do with the text to make this part clearer?

> This is problematic since engineers can potentially waste time implementing commissioning using the Matter framework only to find out later that they must use the Matter Support framework.

You have to use _both_.  You use the Matter Support framework to get an onboarding payload, then you commission via the Matter framework using that payload.

closed 2023-10-19 by woody-apple

maurycyw, 2023-10-19:
@bzbarsky-apple thank you for clarifying. I am new to iOS development and really appreciate your responses (on previous tickets as well). 

The unclarity seems to be around a lack of examples of all of the pieces being put together but also not clearly defining limitations in some of the apis.

bzbarsky-apple, 2023-10-19:
@maurycyw Understood.  That's useful feedback, thank you!