project-chip/connectedhomeip #29874: [QUESTION] RunEventLoop and ScheduleWork can be called without calling InitChipStack?

BurievSardor, 2023-10-19:
Doc says:
    /**
     * InitChipStack() initializes the PlatformManager.  After calling that, a
     * consumer is allowed to call either StartEventLoopTask or RunEventLoop to
     * process pending work.  Calling both is not allowed: it must be one or the
     * other.
     */

    /**
     * ScheduleWork can be called after InitChipStack has been called.  Calls
     * that happen before either StartEventLoopTask or RunEventLoop will queue
     * the work up but that work will NOT run until one of those functions is
     * called.
     *
     * ScheduleWork can be called safely on any thread without locking the
     * stack.  When called from a thread that is not doing the stack work item
     * processing, the callback function may be called (on the work item
     * proc
     * 
But in CHIP Tool example RunEventLoop and ScheduleWork are called without calling InitChipStack:
https://github.com/project-chip/connectedhomeip/blob/349b79525319ad8aac6a901f4afe06234a5a80ac/examples/chip-tool/commands/common/CHIPCommand.cpp#L596

How can it work?
As I see CHip Tool example running without calling InitChipStack.  If it is not so, please correct me.

bzbarsky-apple, 2023-10-20:
> But in CHIP Tool example RunEventLoop and ScheduleWork are called without calling InitChipStack:

```
* thread #1, queue = 'com.apple.main-thread', stop reason = breakpoint 1.1
  * frame #0: 0x00000001014ee2bc chip-tool`chip::DeviceLayer::PlatformManager::InitChipStack(this=0x0000000101c1cae0) at PlatformManager.h:366:9
    frame #1: 0x00000001014eb03c chip-tool`chip::Controller::DeviceControllerFactory::InitSystemState(this=0x0000000101ac59c0, params=FactoryInitParams @ 0x000000016fdfba70) at CHIPDeviceControllerFactory.cpp:119:5
    frame #2: 0x00000001014eaa6c chip-tool`chip::Controller::DeviceControllerFactory::Init(this=0x0000000101ac59c0, params=FactoryInitParams @ 0x000000016fdfc190) at CHIPDeviceControllerFactory.cpp:71:22
    frame #3: 0x0000000100d18acc chip-tool`CHIPCommand::MaybeSetUpStack(this=0x0000000106026880) at CHIPCommand.cpp:129:5
    frame #4: 0x0000000100d1ee7c chip-tool`CHIPCommand::Run(this=0x0000000106026880) at CHIPCommand.cpp:222:5
    frame #5: 0x0000000100dd1834 chip-tool`Commands::RunCommand(this=0x000000016fdff200, argc=5, argv=0x000000016fdff658, interactive=false, interactiveStorageDirectory=0x000000016fdfe900) at Commands.cpp:331:21
    frame #6: 0x0000000100dcf8a0 chip-tool`Commands::Run(this=0x000000016fdff200, argc=5, argv=0x000000016fdff658) at Commands.cpp:178:11
    frame #7: 0x000000010017baf8 chip-tool`main(argc=5, argv=0x000000016fdff658) at main.cpp:52:21
```

So InitChipStack is in fact being called.

bzbarsky-apple, 2023-10-20:
In particular, the link above is to code in `StartWaiting()`, but the callstack I pasted is under `MaybeSetUpStack()`, which is called before `StartWaiting()`.

closed 2023-10-20 by bzbarsky-apple