project-chip/connectedhomeip #29876: [CERT-TEST-FAILURE] - TC-BR-4 - Test failed

KishokG, 2023-10-20:
### Feature Area

Other

### Test Case

TC-BR-4

### Reproduction steps

1. Bring up the DUT with bridge-app.
2. Pair the DUT using chip-tool through "onnetwork" transport.
3. Run the command `./chip-tool powersource read bat-charge-level 1 6`

**Note:** Tried with endpoint 0 to 20.

**Expected Outcome:**

> [1663925822.845859][1588:1588] CHIP:DMG: Endpoint = 0x9,
> [1663925822.845888][1588:1588] CHIP:DMG: Cluster = 0x2f,
> [1663925822.845919][1588:1588] CHIP:DMG: Attribute = 0x0000_000E,
> [1663925822.845946][1588:1588] CHIP:DMG: }
> [1663925822.845978][1588:1588] CHIP:DMG:
> [1663925822.846007][1588:1588] CHIP:DMG: Data = 58,
> [1663925822.846035][1588:1588] CHIP:DMG: },

**Actual Outcome:**

> [1697695642.407312][2131:2133] CHIP:DMG: 					Endpoint = 0x6,
> [1697695642.407380][2131:2133] CHIP:DMG: 					Cluster = 0x2f,
> [1697695642.407450][2131:2133] CHIP:DMG: 					Attribute = 0x0000_000E,
> [1697695642.407514][2131:2133] CHIP:DMG: 				}
> [1697695642.407588][2131:2133] CHIP:DMG: 					
> [1697695642.407641][2131:2133] CHIP:DMG: 				StatusIB =
> [1697695642.407677][2131:2133] CHIP:DMG: 				{
> [1697695642.407712][2131:2133] CHIP:DMG: 					status = 0x7f (UNSUPPORTED_ENDPOINT),
> [1697695642.407744][2131:2133] CHIP:DMG: 				},

### Bug prevalence

Whenever I try to read BatteryChargeLevel attribute.

### GitHub hash of the SDK that was being used

e6f0134483825ce32fd1a9fa86461950f1728231

### Platform

raspi

### Anything else?

Seems it broke after the commit: 52225ddbe6c1587d30afbb3ab23d8a60b45bf2d6

Emill, 2024-02-23:
58 is not even a valid value of BatChargeLevelEnum. Available are 0=OK, 1=Warning, 2=Critical.

KishokG, 2024-07-29:
Issue fixed in https://github.com/project-chip/connectedhomeip/pull/34499/

closed 2024-07-29 by KishokG