project-chip/connectedhomeip #29891: [Test Failed] How to constrain Fabric control permissions with the help of ACLs?

whqxbs, 2023-10-20:
### Test issue(s)

You can see that the matter device has 4 fabrics, each of which has full access to that matter device.
[fabrics.txt](https://github.com/project-chip/connectedhomeip/files/13058035/fabrics.txt)

I wanted to constrain the permissions of certain fabrics with the help of ACL, so I used the following command (assuming that I want to constrain the Fabrics with nodeID 4294967296(You can find it in the .txt file before.) from using the 0x003C Administrator Commissioning Cluster, thus realizing that I won't allow it to open the pairing window):
`./chip-tool accesscontrol write acl '[{"fabricIndex": 0, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, {"fabricIndex": 0, "privilege": 1, "authMode": 2, "subjects": [4294967296], "targets": [{"cluster": 60, "endpoint": null, "deviceType": null}]}]' 1020 0`
But it doesn't work, nodeID 4294967296 still can use the 0x003C Administrator Commissioning Cluster. I'm wondering if I am misunderstanding and making a mistake with the usage of ACLs. If so, How to constrain Fabric control permissions with the help of ACLs?

Thank you!

bzbarsky-apple, 2023-10-21:
ACLs are per-fabric.  When you write ACLs, you only write them on _your own fabric_.  Any entries for other fabrics are not affected.

Node IDs are scoped per fabric.  It's possible to have nodes with id 4294967296 on two different fabrics.

So if you write that on fabric A node ID 4294967296 has no permissions, that does not affect the permissions of node ID 4294967296 on fabric B.

If you need to change the ACLs for a fabric, you need to change them from inside that fabric, by having that fabric issue an ACL write.

closed 2023-10-21 by bzbarsky-apple

whqxbs, 2023-10-21:
You say I need to change them from inside that fabric if I need to change the ACLs for a fabric. As I know, when devices has been paired, one fabric A can easily unpair the other fabric B. 
Does this mean that fabric A has some control over fabric B as well? Is there any way for fabric A to restrict fabric B's access?

bzbarsky-apple, 2023-10-21:
> one fabric A can easily unpair the other fabric B

Correct.

> Does this mean that fabric A has some control over fabric B as well?

The only control fabrics have over each other is being able to remove each other.

> Is there any way for fabric A to restrict fabric B's access?

Apart from removing B entirely, no.