project-chip/connectedhomeip #29907: [Tooling] chip-tool: websocket, why i receive all messages at the same time?

jaggerwoo, 2023-10-23:
### Reproduction steps / Feature

I want to get the progress of the device addition by websocket. I implemented a method of websocket send, and i can receive the message in websocket client, but multiple messages coming in at the same time, why?

1. add a websocket instance class
``` c
// examples/chip-tool/commands/interactive/InteractiveCommands.h
class InteractiveWsInstance
{
public:
    CHIP_ERROR WsSend(const char * msg);
    InteractiveServerCommand * iServer;
};

// Usage:
// 1. #include "../interactive/InteractiveCommands.h"
// 2. gInteractiveWsInstance.WsSend("");
extern InteractiveWsInstance gInteractiveWsInstance;

// examples/chip-tool/commands/interactive/InteractiveCommands.cpp
CHIP_ERROR InteractiveServerCommand::WsSend(const char * msg)
{
    mWebSocketServer.Send(msg);
    return CHIP_NO_ERROR;
}

CHIP_ERROR InteractiveWsInstance::WsSend(const char * msg)
{
    iServer->WsSend(msg);
    return CHIP_NO_ERROR;
}
```

2. i want send a progress 1 when add device
``` c
void PairingCommand::OnPairingComplete(CHIP_ERROR err)
{
    if (err == CHIP_NO_ERROR)
    {
        ChipLogProgress(chipTool, "Pairing Success");
        ChipLogProgress(chipTool, "PASE establishment successful");
        std::stringstream nodeID;
        char nodeIDBuf[CHIP_CONFIG_LOG_MESSAGE_MAX_SIZE];
        snprintf(nodeIDBuf, sizeof(nodeIDBuf), "0x%" PRIx64, mNodeId);
        nodeID << nodeIDBuf;
        std::stringstream content;
        content << "{\"queue\":\"devices\",";
        content << "\"node_id\":\"" << nodeID.str() << "\",";
        content << "\"op_state\":\"in_progress_1\"}";
        gInteractiveWsInstance.WsSend(content.str().c_str());
        ChipLogProgress(chipTool, "1111----------%" PRIu64, mNodeId);
        if (mPairingMode == PairingMode::CodePaseOnly || mPaseOnly.ValueOr(false))
        {
            SetCommandExitStatus(err);
        }
    }
    else
    {
        ChipLogProgress(chipTool, "Pairing Failure: %s", ErrorStr(err));
    }

    if (err != CHIP_NO_ERROR)
    {
        SetCommandExitStatus(err);
    }
}
```

3. send a progress 2 when add device
``` c
void PairingCommand::OnStatusUpdate(DevicePairingDelegate::Status status)
{
    switch (status)
    {
        case DevicePairingDelegate::Status::SecurePairingSuccess:{
            ChipLogProgress(chipTool, "Secure Pairing Success");
            ChipLogProgress(chipTool, "CASE establishment successful");

            std::stringstream nodeID;
            char nodeIDBuf[CHIP_CONFIG_LOG_MESSAGE_MAX_SIZE];
            snprintf(nodeIDBuf, sizeof(nodeIDBuf), "0x%" PRIx64, mNodeId);
            nodeID << nodeIDBuf;

            std::stringstream content;
            content << "{\"queue\":\"devices\",";
            content << "\"node_id\":\"" << nodeID.str() << "\",";
            content << "\"op_state\":\"in_progress_2\"}";
            gInteractiveWsInstance.WsSend(content.str().c_str());
            ChipLogProgress(chipTool, "2222----------%" PRIu64, mNodeId);

            break;
        }
        case DevicePairingDelegate::Status::SecurePairingFailed: {

            ChipLogError(chipTool, "Secure Pairing Failed");
            SetCommandExitStatus(CHIP_ERROR_INCORRECT_STATE);
            break;
        }
    }
}
```

so i send two diff messages in different time, but i receive these two messages at the same time, why? how can i solve it
[image]

### Platform

raspi

### Type

Platform validated

bzbarsky-apple, 2023-10-23:
> so i send two diff messages in different time

The in_progress_2 message is sent when AddNOC succeeds (note that the logs above are wrong: CASE has _not_ been established at this point).

It's _possible_ that the time from "PASE established" to "AddNOC succeeded" was in fact quite short, depending on the device being commissioned; nothing in the logs above indicates how much time actually passed between the notifications.

jaggerwoo, 2023-10-24:
> It's _possible_ that the time from "PASE established" to "AddNOC succeeded" was in fact quite short, depending on the device being commissioned; nothing in the logs above indicates how much time actually passed between the notifications.

But on the websocket server log, it does look like a different time;
1. in_progress_1, i print 111
``` json
[1698114104.766412][68592:68595] CHIP:EM: Found matching exchange: 44677i, Delegate: 0x7f98016ec0
[1698114104.766563][68592:68595] CHIP:SC: SecureSession[0x7f980592a0, LSID:11917]: State change 'kEstablishing' --> 'kActive'
[1698114104.766614][68592:68595] CHIP:IN: SecureSession[0x7f980592a0]: Activated - Type:1 LSID:11917
[1698114104.766654][68592:68595] CHIP:IN: New secure session activated for device <FFFFFFFB00000000, 0>, LSID:11917 PSID:41840!
[1698114104.766704][68592:68595] CHIP:CTL: Remote device completed SPAKE2+ handshake
[1698114104.766742][68592:68595] CHIP:TOO: Pairing Success
[1698114104.766776][68592:68595] CHIP:TOO: PASE establishment successful
[1698114104.766850][68592:68595] CHIP:TOO: 1111----------199
[1698114104.766896][68592:68595] CHIP:CTL: Commissioning stage next step: 'SecurePairing' -> 'ReadCommissioningInfo'
[1698114104.766939][68592:68595] CHIP:CTL: Performing next commissioning step 'ReadCommissioningInfo'
[1698114104.766976][68592:68595] CHIP:CTL: Sending request for commissioning information
[1698114104.767038][68592:68595] CHIP:DMG: SendReadRequest ReadClient[0x7f98001280]: Sending Read Request
```
2. in_progress_2, i print 222
``` json
[1698114111.885679][68592:68595] CHIP:DMG: Received Command Response Data, Endpoint=0 Cluster=0x0000_003E Command=0x0000_0008
[1698114111.885737][68592:68595] CHIP:CTL: Device returned status 0 on receiving the NOC
[1698114111.885775][68592:68595] CHIP:CTL: Operational credentials provisioned on device 0x7f98016e70
[1698114111.885811][68592:68595] CHIP:TOO: Secure Pairing Success
[1698114111.885841][68592:68595] CHIP:TOO: CASE establishment successful
[1698114111.885909][68592:68595] CHIP:TOO: 2222----------199
[1698114111.885958][68592:68595] CHIP:CTL: Successfully finished commissioning step 'SendNOC'
[1698114111.885996][68592:68595] CHIP:CTL: No NetworkScan enabled or WiFi/Thread endpoint not specified, skipping ScanNetworks
[1698114111.886031][68592:68595] CHIP:CTL: Commissioning stage next step: 'SendNOC' -> 'WiFiNetworkSetup'
```

bzbarsky-apple, 2023-10-24:
Sounds like there might be buffering going on somewhere here, perhaps...